juj/fbcp-ili9341 #85: Portrait mode on ili9341 screen

lostinspacebar, 2019-05-30:
Hello,

I'm trying to show emulation station in portrait mode. I've changed /boot/config.txt to have display_rotate=1.

When I run fbcp-ili9341, this is the output I get (truncated for brevity):
`
...
Relevant source display area size with overscan cropped away: 480x640.
Source GPU display is 480x640. Output SPI display is 320x240 with a drawable area of 320x240. Applying scaling factor horiz=0.38x & vert=0.38x, xOffset: 70, yOffset: 0, scaledWidth: 180, scaledHeight: 240
Creating dispmanX resource of size 180x240 (aspect ratio=0.750000).
GPU grab rectangle is offset x=0,y=0, size w=180xh=240, aspect ratio=0.750000
...
`
So the HDMI output has been rotated as per config.txt but the SPI driver is still trying to draw landscape. And that's basically what I see on screen:
[image]

I've tried disabling #define DISPLAY_OUTPUT_LANDSCAPE in config.h and then the console output shows that the SPI driver is also drawing to a 240x320 surface but the actual output on screen is just landscape but mirrored. 

What else can I fiddle with to get this to actually show portrait mode (4:3) on this LCD?

Thanks,
Aditya

juj, 2019-05-30:
> I've tried disabling #define DISPLAY_OUTPUT_LANDSCAPE in config.h and then the console output shows that the SPI driver is also drawing to a 240x320 surface but the actual output on screen is just landscape but mirrored.

Disabling `#define DISPLAY_OUTPUT_LANDSCAPE` is supposed to do what you are looking for. However you have combined that with `display_rotate=1` - I think that is what is throwing off fbcp-ili9341, it does not know that the HDMI output is rotated and it should "unrotate" it first before applying its own rotation. Basically it is expecting that `display_rotate=0` is in effect.

You can customize the rotation in more detail by adjusting the MADCTL register directly. That is set in this line

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/ili9341.cpp#L52

(assuming you are on ILI9341 controller. Other controller files have an equivalent MADCTL line in them)

You can try e.g. changing

```
#define MADCTL_ROTATE_180_DEGREES (MADCTL_COLUMN_ADDRESS_ORDER_SWAP | MADCTL_ROW_ADDRESS_ORDER_SWAP)
```
to
```
#define MADCTL_ROTATE_180_DEGREES (MADCTL_COLUMN_ADDRESS_ORDER_SWAP | MADCTL_ROW_COLUMN_EXCHANGE)
```
or
```
#define MADCTL_ROTATE_180_DEGREES (MADCTL_ROW_COLUMN_EXCHANGE | MADCTL_ROW_COLUMN_EXCHANGE)
```

lostinspacebar, 2019-05-30:
So if I disable DISPLAY_OUTPUT_LANDSCAPE, wouldn't the source HDMI output still be a 4:3 aspect (320x240)? I want the output to be 3:4 (think vertical games in emulators etc.).

I guess I'll play with those two settings and the MADCTL lines and report back.

Thanks,
Aditya

juj, 2019-05-31:
> So if I just disable DISPLAY_OUTPUT_LANDSCAPE without the display_rotate=1 in config.txt, wouldn't the source HDMI output still be a 4:3 aspect (320x240)?

I think the way the resolution & HDMI rotation works is that you set the resolution in /boot/config.txt, and fbcp-ili9341 always sees that resolution as if display_rotate=0 was in effect, even if some other setting had been set. Setting display_rotate=1 only applies to HDMI output (which I presume will output 240x320 picture stream if /boot/config.txt had resolution of 320x240 set - not 100% sure on that though)

When emulators play a vertical game in a regular 320x240 display resolution, do they pillarbox the contents off from left and right if the game is 240x320? Or do the emulators render sideways?

If they render it top-down with pillarboxes, then I suppose such games would work ok by setting /boot/config.txt resolution of 240x320 and display_rotate=0 and disable `#define DISPLAY_OUTPUT_LANDSCAPE` with no settings to MADCTL line.

The MADCTL register is able to perform arbitrary kind of rotation and mirroring, so should be able to use that to get what you want. However if there are pillarboxes/letterboxes present in the source video, you may need to apply custom cropping to fbcp-ili9341's output, which can be done via #defines as seen in https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/freeplaytech_waveshare32b.h#L19 . However such cropping may be a bit unideal, since it applies to overall video in all cases, so non-game screens also will get the crop. It will be best if the /boot/config.txt resolution is configured in a way that avoids the game needing to render pillarboxes/letterboxes.

lostinspacebar, 2019-05-31:
Alright! I've got it like I wanted. 

What I had to do:

/boot/config.txt -- Make these settings the following values.
`
hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 640 60 1 0 0 0
hdmi_force_hotplug=1
`

config.h: Disable #define DISPLAY_OUTPUT_LANDSCAPE by commenting it out

ili9341.cpp: Add the following line:
`madctl ^= MADCTL_COLUMN_ADDRESS_ORDER_SWAP;`
right before the line 
`SPI_TRANSFER(0x36/*MADCTL: Memory Access Control*/, madctl);`

This gives me what I wanted. Some games which are made to be played in landscape still show up with black bars but emulation station itself fills up nicely. Haven't tried arcade emulators yet.

Hope this helps other peeps. @juj Thank you so much for the detailed info and for this awesome piece of software!

juj, 2019-07-23:
Closing resolved issues.

closed 2019-07-23 by juj