juj/fbcp-ili9341 #86: Can't get ILI9341 screen to work! Please help!

Melonhead60, 2019-06-06:
I've been trying to get [this ](https://www.aliexpress.com/item/32845973656.html?spm=2114.search0104.3.9.700e6c3dcl8wtl&ws_ab_test=searchweb0_0%2Csearchweb201602_5_10065_10068_10130_5731315_318_10890_5730315_10547_319_10546_10548_317_10545_5728815_10696_10084_10083_10618_452_5729215_10307_532_5731115_5731415_5731215_5731615_204_328_10059_10884_5731515_323_325_10887_100031_320_321_322_10103_5731715_5728415-10890%2Csearchweb201603_55%2CppcSwitch_0&algo_expid=5d3eabb3-06ba-4d9b-ad37-f929f18680ac-1&algo_pvid=5d3eabb3-06ba-4d9b-ad37-f929f18680ac&transAbTest=ae803_4) screen to work with this driver for about a week now and i'm at my wits end. I keep running into problems faster then I can fix them. When it's not white screening after 5 seconds, it's displaying a garbled mess. When it's not doing either of those, it's just not updating half the pixels on the screen (or stops updating pixels at all). I've supplied better power, I've tried changing the DIVISOR to every multiple of 2 from 2 up to 200, I've tried every solution found on this forum, and I've redone and rechecked my soldering job more times then I can count. At this point I'm just curious if this screen is compatible with the code at all, so if anyone wants to take a stab at getting it to work PLEASE let me know if you can.

Plus, this screen would be AMAZING with this driver. 2.8 inches, 60fps, AND for only 6$?  Sign me up!

juj, 2019-06-06:
ILI9341 should be well compatible with this driver. You can try using ILI9340 driver code path instead, which is very similar, but works to avoid some initialization time incompatibilities that were observed in #47. (If trying that out, use a new build directory or delete the previous intermediate build directory, or the CMakeCache.txt file in it before reconfiguring to a different driver)

Also you can try using the built-in Linux fbtft+spi driver for the display - if that works, then it suggests it could be some incompatibility in the fbcp-ili9341 initialization sequence for the particular display. You can try commenting out some of the initialization directives at https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/ili9341.cpp#L34 . Especially those that affect power, vcom or timing might have an impact here. When switching between different init directives in this manner, it is best to hard power off the display and then reconnect it between attempts so that the internal registers of the display get cleared for each test run.

Melonhead60, 2019-06-06:
Thanks for the quick response! I'll try that, and get back to you :)

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj