juj/fbcp-ili9341 #88: Aspect ratio issue with waveshare 3.2"

riv-er, 2019-06-27:
Firstly, really awesome project you have going down here. I have been using the 3.2" waveshare display with your driver and it has been pretty fast and usable enough. The problem is that a bit of the top and bottom of the screen cannot be used at all, I tried playing around with config.txt with all weird configurations even tried a bunch of options in the cmake file but still could not figure it out. It seems it is not being whatsoever. I gave some really resolutions but it still wouldn't occupy those spaces

juj, 2019-06-28:
Hmm, perhaps you are using the Freeplaytech profile for the Waveshare 3.2"? Then that has been adjusted to apply to the visible geometry of the Freeplaytech CM3 handheld device, where parts of the display are obstructed by the handheld device's bezels. To adjust for bare display, edit

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/freeplaytech_waveshare32b.h#L19-L32

and change the numbers on all the four lines to 0.

Alternatively, you can try with the generic `-DILI9341=ON` profile instead of the `-DFREEPLAYTECH_WAVESHARE32B=ON` profile in CMake (Waveshare 3.2" is an ILI9341), which avoids the geometric change.

If that is not it, then give attention to the resolution numbers printed at fbcp-ili9341 startup. That will show the amount of aspect ratio scaling that will be performed. Based on that info, try adjusting the resolution in /boot/config.txt to match a multiple of the display resolution 320x240. (e.g. set at 320x240 or 640x480)

riv-er, 2019-07-01:
Thanks for the quick and timely reply, sorry for my late one. It was not actually your driver problem apparently I some how messed up the cmdline and the preferred resolution was not getting applied to my window which had the this driver running, basically whenever I changed the resolution in config.txt, it made no change. 
Thanks a lot again for your work and sorry for bothering you

closed 2019-07-01 by riv-er