juj/fbcp-ili9341 #89: Waveshare 3.5 Pi2B+ Black screen

mailmartinviljoen, 2019-06-28:
I built the config using the following configuration
sudo cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DWAVESHARE35B_ILI9486=ON -DARMV7A=ON ..

Made sure I killed the fcb driver, 
Below is my boot config
dtparam=audio=on
hdmi_force_hotplug=1
#max_usb_current=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 640 480 60 6 0 0 0
hdmi_drive=2
display_rotate=0

Also tried 
hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1

after building and making the config it all seems to compile just fine with no errors.
When I then execute the driver the screen just stays black.

If i put the original LCD35B-show-V2 driver back reboot, retro pi shows up fine. If I then execute the fbcp-lil.. driver whilst the normal driver is running the screen turns off. So that atleast confirms that something is happening.

If you know what the issue could be I would GREATLY appreciate it.

mailmartinviljoen, 2019-06-28:
Nevermind I removed all settings in the /boot/config.txt file and replaced it with 

hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1

juj, 2019-07-20:
In #82 @Mootikins identified a problem that prevented ILI9341 displays from starting up in some cases. It is possible that the issue would affect ILI9486 as well. A fix was pushed in at https://github.com/juj/fbcp-ili9341/commit/b2eeec1de91dc6967e3331c37dcec0fcecb1b8b0, so you `git pull` or reclone to get the fix, may be worth a shot.

Though this reads like you were already able to resolve the issue, so I'll close this one up.

closed 2019-07-20 by juj