juj/fbcp-ili9341 #91: 320x240 ST7789VW screen - can't get resolution higher than 240x240

specializationisoverrated, 2019-07-13:
Hi, I've got a 320x240 ST7789VW screen that I'm trying to work with. Its wired up and working with the DST7789VW option but for some reason I can't get it to display a resolution higher than 240x240. Changing the display size in boot/config.txt doesn't seem to work. 

The output text when running the driver acknowledges the issue by saying: "Source GPU display is 320x240. Output SPI display is 240x240 with a drawable area of 240x240. Applying scaling factor...." Not sure which option or flag I should use to force a resolution or is this built into the ST7789VW driver since the 2 popular screens with this controller have a square aspect ratio?

specializationisoverrated, 2019-07-13:
I was able to solve the problem above. Took me a moment to realize that you had put ST7789 in with the ST7735r.h file. Made the changes there and got the screen to work. This led to a different issue but I'll open up a separate thread for that.

closed 2019-07-13 by specializationisoverrated

specializationisoverrated, 2019-07-13:
One more note on this. For those of you using a 320x240 ST7789 or ST7789VW make sure you remember to comment out or change the +Y=80 value in st7735r.cpp otherwise the screen output will be shifted 80 pixels over.

juj, 2019-07-15:
Thanks for the tip. Perhaps the following patch would avoid having to edit st7735r.cpp directly?

```diff
diff --git a/st7735r.cpp b/st7735r.cpp
index 5bf4811..9ecc9c1 100644
--- a/st7735r.cpp
+++ b/st7735r.cpp
@@ -92,7 +92,7 @@ void InitST7735R()
     // memory in row addresses Y = 319-(0...239) = 319...80 range. To view this range, we must scroll the view by +80 units in Y
     // direction so that contents of Y=80...319 is displayed instead of Y=0...239.
     if ((madctl & MADCTL_ROW_ADDRESS_ORDER_SWAP))
-      SPI_TRANSFER(0x37/*VSCSAD: Vertical Scroll Start Address of RAM*/, 0, 80);
+      SPI_TRANSFER(0x37/*VSCSAD: Vertical Scroll Start Address of RAM*/, 0, 320 - DISPLAY_WIDTH);
 #endif
 ```

then one only needs to edit one location in st7735r.h to change display width to 320 as opposed to 240? If that works well for you, I can commit that in.

specializationisoverrated, 2019-07-22:
Hi, sorry didn't realize you had replied. This fix works for me.

juj, 2019-07-23:
Great to hear, pushed the change, so now changing only one location should set up 320x240 display.

unphased, 2019-08-13:
@specializationisoverrated Could you clarify whether your code change increased DISPLAY_NATIVE_HEIGHT to 320? I have a different display from you (adafruit 320x240 ST7789) and I had to change the DISPLAY_NATIVE_HEIGHT, rather than the DISPLAY_NATIVE_WIDTH to make it work properly. This means that the `320-DISPLAY_NATIVE_WIDTH` logic is still causing problems.

specializationisoverrated, 2019-08-13:
Hey, I'm away for a few days but will check as soon as I can. (sometime this upcoming weekend). Not sure exactly which panel Adafruit is using I just have a random raw panel bought from China. Will keep you posted.

unphased, 2019-08-13:
Last week or two. Actually. 

Is yours a Waveshare item, or is its pedigree even more murky?

specializationisoverrated, 2019-08-13:
Nah, no name panel from Aliexpress, just the panel w/o a breakout board. Working on a quick little retropie project and ultimately need to integrate it into my own PCB to save on space.