juj/fbcp-ili9341 #93: Fbcp-ili9341 not loading on launch

holadivinus, 2019-07-19:
When i run sudo ./fbcp-ili9341 in the build dir, it doesnt do anything. at all. :(

i compiled it with the command 
cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=18 -DSTATISTICS=0 -DUSE_DMA_TRANSFERS=OFF -DDISPLAY_INVERT_COLORS=ON ..

:/

holadivinus, 2019-07-19:
When i compile it as 
cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=18 -DSTATISTICS=0 -DUSE_DMA_TRANSFERS=OFF ..

it works but with inverted colors.

holadivinus, 2019-07-19:
on a waveshare 3.5 inch display (B)

juj, 2019-07-19:
Display color inversion for ILI9486 is managed at https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/ili9486.cpp#L50 . My WaveShare 3.5 B display needed display inversion enabled for correct colors. You can try debugging around that code location (spec sheet can be found at https://cdn.datasheetspdf.com/pdf-down/I/L/I/ILI9486-ILITEK.pdf ) No idea I'm afraid if display inversion does not do it :/

holadivinus, 2019-07-20:
as i said before, inverted colors command just doesnt work. im going to switch back to a waveshare A. ;-;

juj, 2019-07-20:
In #82 @Mootikins identified a problem that prevented ILI9341 displays from starting up in some cases. It is possible that the issue would affect ILI9486 as well. A fix was pushed in at https://github.com/juj/fbcp-ili9341/commit/b2eeec1de91dc6967e3331c37dcec0fcecb1b8b0, so you `git pull` or reclone to get the fix, may be worth a shot.

juj, 2019-07-20:
Might be related - In https://github.com/juj/fbcp-ili9341/issues/83 there is an identified miscoloring on WaveShare 3.5 B display, that the invert colors command does not fix up. So there is some issue with that specific display's color space.

r-trigo, 2020-01-29:
> When i run sudo ./fbcp-ili9341 in the build dir, it doesnt do anything. at all. :(

I am having the same problem :disappointed: 
- RPi 0 W
- Raspbian Lite September 2019 version (also tried with RetroPie)
- ILI9341

Pinout
- Vcc   -> 5V
- GND   -> GND
- CS    -> GPIO 8
- Reset -> GPIO 25
- D/C   -> GPIO 24
- MOSI  -> GPIO 10
- CLK   -> GPIO 11
- LED   -> GPIO 15
- MISO -> GPIO 9

Build command:
`cmake -DILI9341=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_BACKLIGHT=15 ..`

And screen stays black like when not wired.

juj, 2020-12-08:
Closing out old issues. It is worth checking against "official" drivers if fbcp-ili9341 is not working out at all, to double-check the hardware wiring.

closed 2020-12-08 by juj