juj/fbcp-ili9341 #94: Add Raspberry Pi 4 Model B support

juj, 2019-07-19:
It is likely that Pi 4 Model B will not work out of the box due to completely different SoC "BCM2711" vs the older BCM2835, but a full data sheet has not been published. Also it is uncertain if DispmanX API is supported, or if some other API should be used to pull GPU frames.

This issue is to discuss enabling Pi 4 Model B support.

unphased, 2019-08-19:
My initial testing yielded this failure: 

```
bcm_host_get_peripheral_address: (nil), bcm_host_get_peripheral_size: 4261412864, bcm_host_get_sdram_address: 0xc0000000
mapping /dev/mem failed
```

juj, 2019-08-25:
Hmm, testing out on Raspbian Buster on a Pi 4B, it surprisingly worked out of the box with no modifications needed. Printed out
```
pi@raspberrypi:~/fbcp-ili9341/build $ sudo ./fbcp-ili9341 
bcm_host_get_peripheral_address: 0xfe000000, bcm_host_get_peripheral_size: 25165824, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 250000000hz, max turbo: 500000000hz. SPI CDIV: 6, SPI max frequency: 83333333hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0xb530f000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0xb530f700, DMA RX channel register file is at ptr: 0xb530f100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 25
Creating SPI task thread
InitSPI done
DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
Relevant source display area size with overscan cropped away: 320x480.
Source GPU display is 320x480. Output SPI display is 320x480 with a drawable area of 320x480. Applying scaling factor horiz=1.00x & vert=1.00x, xOffset: 0, yOffset: 0, scaledWidth: 320, scaledHeight: 480
Creating dispmanX resource of size 320x480 (aspect ratio=0.666667).
GPU grab rectangle is offset x=0,y=0, size w=320xh=480, aspect ratio=0.666667
All initialized, now running main loop...
```
at startup.

That confirms that the new Pi 4B SPI/GPIO address space is compatible with Pi 3B, so it looks like there are no special changes to accommodate to here - closing this as resolved.

closed 2019-08-25 by juj

juj, 2019-08-25:
On further testing, it looks like there are some performance oddities with the WaveShare 3.5" C display I am testing with Raspberry Pi 4 on Buster: running the display at speed 300/4=75MHz works, and also speed 500/6=83.33MHz works, but speed 310/4=77.5MHz does not. Have not seen this type of effect present before on earlier Pis/displays, not sure what is going on there.