juj/fbcp-ili9341 #95: Can fbcp-ili9341 use SPI0 on pins 35-39 on a Compute Module?

MichaelRColton, 2019-07-21:
I'm designing a Compute Module based device with a screen and would like to use fbcp-ili9341 to drive the display, but I don't want to touch the normal pins on the the 40 pin connector (because I still want that to be available for hats)

The Compute Modules expose more GPIO than the normal Pis. I looks like SPI0 is available on pins 35-39 (Alt 0... but the normal SPI0 is also Alt 0... not sure how that works yet)

Will fbcp-ili9341 work with those pins? I'd check myself but I don't have the IO board and haven't finished designing my own board yet.

Thanks!!

juj, 2019-07-21:
I am not familiar with the Compute Module to say how it works. It is possible that the only needed change is to adjust to use different pins in https://github.com/juj/fbcp-ili9341/blob/b2eeec1de91dc6967e3331c37dcec0fcecb1b8b0/spi.h#L51-L55 though if the pin numbers exceed 31, then also the following need changing:

https://github.com/juj/fbcp-ili9341/blob/b2eeec1de91dc6967e3331c37dcec0fcecb1b8b0/spi.h#L70-L72

with something like

```c
#define GET_HIGH_GPIO(pin) (gpio->gplev[1] & (1 << (pin))) // Pin must be (32-63)
#define SET_HIGH_GPIO(pin) gpio->gpset[1] = 1 << (pin) // Pin must be (32-63)
#define CLEAR_HIGH_GPIO(pin) gpio->gpclr[1] = 1 << (pin) // Pin must be (32-63)
```
if I remember correctly - then call the appropriate macro in places that manipulate the pins. (You can also merge the two register file accesses to one macro that checks between the pin number, though that will bring some performance penalty)

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj