juj/fbcp-ili9341 #96: OMX Player crashes driver [I think]

SlowCatalyst, 2019-07-25:
Hi there- was just seeing if perhaps the specifics could be found as to why OMX might be crashing the driver.

I am using the FreePlay CM3, and when using this driver, it seems that any instance that OMX is called up- be it video splash, game list videos in RetroPie [using OMX], or video screensaver [using OMX]- it causes a black screen, and hangs until I SSH and restart the driver, but will crash again anytime OMX for videos is called up once more.

Any idea what I could be doing wrong?

This is what my settings are when building the driver:
-DARMV8A=ON -DFREEPLAYTECH_WAVESHARE32B=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON -DUSE_DMA_TRANSFERS=ON -DDMA_TX_CHANNEL=1 -DDMA_RX_CHANNEL=5 -DSTATISTICS=0 ..

juj, 2019-07-26:
I think the root cause is the same as #28 - a bug in DispmanX causes it to stop functioning when screen is resized. The only known workaround has been to implement a reload daemon to spin fbcp-ili9341 back up when it crashes.

Mootikins, 2019-08-13:
This may not be directly related to the DispmanX resize bug since it happens when you are using OMXPlayer to play scraped video in EmulationStation, while our old driver (rpi-fbcp) also uses DispmanX and has none of these issues.

SlowCatalyst, 2019-08-22:
Found the conflict!

Thanks for the input, fellas.
I found that simply changing my cmake options from the above example I provided, to:

-DARMV8A=ON -DFREEPLAYTECH_WAVESHARE32B=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON -DUSE_DMA_TRANSFERS=ON -DDMA_TX_CHANNEL=7 -DDMA_RX_CHANNEL=5 -DSTATISTICS=0 ..

So, "DDMA_TX_CHANNEL=7"

Videos with OMX are now playing! Video splash, video screensaver, and gamelist videos, all!
Man, with this driver, it's all so crisp now!

Thanks for all of your work, juj and Mootikins!

closed 2020-12-08 by juj