juj/fbcp-ili9341 #99: #define ALL_TASKS_SHOULD_DMA

phansen39, 2019-07-28:
/home/pi/fbcp-ili9341/spi.cpp: In function ‘void RunSPITask(SPITask*)’:
/home/pi/fbcp-ili9341/spi.cpp:281:24: error: ‘SPIDMATransfer’ was not declared in this scope
     SPIDMATransfer(task);
                        ^
/home/pi/fbcp-ili9341/spi.cpp:288:26: error: ‘WaitForDMAFinished’ was not declared in this scope
       WaitForDMAFinished();

juj, 2019-07-28:
the two functions are declared in https://github.com/juj/fbcp-ili9341/blob/aa01d12cd78cd15fd655a96b4811eb9437da136a/dma.h#L127-L135

conditional to `USE_DMA_TRANSFERS` being defined. `spi.cpp` does include that file: https://github.com/juj/fbcp-ili9341/blob/aa01d12cd78cd15fd655a96b4811eb9437da136a/spi.cpp#L13

so this suggests that you were having a build where `ALL_TASKS_SHOULD_DMA` was defined, but `USE_DMA_TRANSFERS` was not. Pushed a commit to verify that code should not be getting compiled with such mismatching directives. Run a `git pull` and double check build flags?

phansen39, 2019-07-31:
Got it thankyou juj

closed 2019-07-31 by phansen39