juj/fbcp-ili9341 #105: Panel from BuyDisplay - White screen

h3r0n, 2019-08-20:
I just tried [this](https://www.buydisplay.com/default/serial-spi-3-2-inch-tft-lcd-module-display-ili9341-power-than-sainsmart) panel from BuyDisplay (which is among the tested displays).
Despite the output to fbcp-ili9341 seems to be happy, the display just stays white.

```
bcm_host_get_peripheral_address: 0x20000000, bcm_host_get_peripheral_size: 33554432, bcm_host_get_sdram_address: 0x40000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0xb4b21000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0xb4b21700, DMA RX channel register file is at ptr: 0xb4b21100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 24
InitSPI done
Relevant source display area size with overscan cropped away: 720x480.
Source GPU display is 720x480. Output SPI display is 320x240 with a drawable area of 320x240. Applying scaling factor horiz=0.44x & vert=0.44x, xOffset: 0, yOffset: 13, scaledWidth: 320, scaledHeight: 213
Creating dispmanX resource of size 320x213 (aspect ratio=1.502347).
GPU grab rectangle is offset x=0,y=0, size w=320xh=213, aspect ratio=1.502347
All initialized, now running main l
```

Setup:
Raspberry Pi Zero W, Raspbian Stretch
[image]

rafistolique, 2019-08-20:
Hi to you,

I will start by saying that I am not an expert as many are here. You'll probably have a better answer than mine. I must admit I can not check your wiring, I do not know anything. I will surely say nonsense.

But assuming that you have plugged all the cables, I see that your driver is the same as in my case. And if I can afford to suggest something is that in my case (in gpio) if the screen remains blank after using: sudo ./fbcp-ili9341

It's either because of a bad choice of driver, or, and the most plausible is that your "-DSPI_BUS_CLOCK_DIVISOR = 6" is configured too hard (if you let the speed of the core on its configuration of origin).
Question very stupid but I apologize and have to ask, did you test with a less difficult divisor? kind: -DSPI_BUS_CLOCK_DIVISOR = 24?

Sorry in advance if my answer seems stupid, I try to help, but in the worst case you will have to wait for someone more competent than me ...

Good luck and sorry for my english ...

h3r0n, 2019-08-20:
Thank you. Unfortunately I just tried with 24 and I still get a white screen.

rafistolique, 2019-08-20:
Arfff.

I tried, sorry it did not work.

Personally I had started at 30 to go down to 18, then I had to use: -DUSE_DMA_TRANSFERS = OFF to go lower ...

quezadaminter, 2020-06-22:
@h3r0n is this the 320x480 display from BuyDisplay.com? We just got that display with the ili9488 controller working.

h3r0n, 2020-06-22:
> @h3r0n is this the 320x480 display from BuyDisplay.com? We just got that display with the ili9488 controller working.

Yes it is. I'll try again asap. Do I need to change anything to my configuration?

quezadaminter, 2020-06-22:
When we were trying to get he 9488 working, another user found that it was possible to talk to it with the 9486L driver but hte colors were wrong. So we used the 9486L as a basis, the only think we changed were the initialization commands. In researching the initialization commands we found a lot of overlap between the models. I guess my point is that it may be able to work if you try different combinations. Definitely try raising the clock divider so you start with a slow bus speed and try turning the DMA off. Also, if you can, I found the datasheet for the 9488 on BuyDisplay.com's website, you may want to grab that and use it to compare the initialization instructions, maybe you find a clue.

Good luck!

juj, 2020-12-08:
Sorry, missed this thread. Closing out old issues, feel free to reopen or create a new thread if still struggling with a problem.

closed 2020-12-08 by juj