juj/fbcp-ili9341 #108: Only white screen

gapdii, 2019-08-29:
I bought a display at the link below.

https://www.aliexpress.com/item/32856132907.html?spm=a2g0s.9042311.0.0.27424c4dJhHiow

There was no CS, LED and only 8pin.
(Gnd, Vcc, Clk, Mosi, Res, DC, Blk, Miso)

I want to know how to connect to the raspberry and enter CMAKE commands.

 am sorry that I am not good at English.

juj, 2019-08-29:
The display specs list

Driver IC: ILI9341

Interface:
GND (power ground)	
VCC (power supply is 3.3V)
CLK (clock line)
MOSI (data input)
RES (reset line)
DC (data/command selection line)
BLK (backlight)
MISO (data output) 

so connect the pins to hardware SPI0. See https://pinout.xyz/# where the pink pins are SPI0 (BCM9, BCM10 and BCM11), and use the ILI9341 CMake directive option from https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself .

juj, 2019-08-29:
For the RES, DC and BLK, you can connect them to any free general GPIO pins, like BCM 23, BCM 24 and BCM 25, but remember to configure them in CMake at https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself

gapdii, 2019-08-29:
> For the RES, DC and BLK, you can connect them to any free general GPIO pins, like BCM 23, BCM 24 and BCM 25, but remember to configure them in CMake at https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself

Thanks. I'll try this!

gapdii, 2019-08-29:
> For the RES, DC and BLK, you can connect them to any free general GPIO pins, like BCM 23, BCM 24 and BCM 25, but remember to configure them in CMake at https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself

You did it!
Thank you very much.
[image]

juj, 2019-08-29:
Great job!

closed 2019-08-29 by juj