juj/fbcp-ili9341 #109: error: '-mfloat-abi=hard': selected processor lacks an FPU

billthecatt, 2019-09-03:
Building on an RPiZero using:

 cmake -DCMAKE_VERBOSE_MAKEFILE=ON -DWAVESHARE_ST7735S_HAT=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DSTATISTICS=OFF -DUSE_DMA_TRANSFERS=OFF -DSINGLE_CORE_BOARD=ON  -DARMV6Z=ON ..

-- Doing a Release build
-- Board revision: 9000c1
-- Detected this Pi to be one of: Pi A, A+, B rev. 1, B rev. 2, B+, CM1, Zero or Zero W, with single hardware core and ARMv6Z instruction set CPU.
-- Targeting a Raspberry Pi with only one hardware core
<snip>
-- Targeting WaveShare 128x128 1.44inch LCD Hat with ST7735S controller
-- Configuring done
-- Generating done
-- Build files have been written to: /root/fbcp-ili9341/build

make -j

Things are building ok until:
[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o -c /root/fbcp-ili9341/mem_alloc.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o -c /root/fbcp-ili9341/statistics.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o -c /root/fbcp-ili9341/ssd1351.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/text.cpp.o -c /root/fbcp-ili9341/text.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o -c /root/fbcp-ili9341/st7735r.cpp
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:118: CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o] Error 1

Pretty sure the detection code in cmake is working ok, not sure why it's barfing bout lacking an FPU.

Any ideas how I can make it past this build error?

juj, 2019-09-04:
Raspi zero should definitely support hard float abi. But if that is giving grief, you can try removing the directive from line https://github.com/juj/fbcp-ili9341/blob/7fd16bf60a4f87a7e497fe38e6862c50132f70f5/CMakeLists.txt#L49 . Delete CMakeCache.txt after editing the line for good measure, sometimes old options linger around in the cache file.

juj, 2019-09-04:
Btw for raspi zero DMA will be critical, otherwise the driver will consume so much cpu that other processes starve. (Disabling DMA would require lowering display update FPS to 5 or 10 refreshes per second)

billthecatt, 2019-09-04:
Thanks so much for the awesome project and help w/ the hard float abi settings. 

I hear you that "dma will be critical" but I don't know what to do with that information. Sort of feels like the documentation or the configuration script should be detecting raspi zero and then setting DMA to something appropriate rather than just leaving it to end user confusion. I don't know what the correct setting here is, which is why I took a stab at something. Could you perhaps suggest a range that's valid? I get the feeling from your suggestion that mine is wrong, but without your experience i don't even know if I should be setting it higher or lower.. I'm all ears tho.. what's the right thing here? ( Suggest the script and/or documentation detail what it should be, sorry if I missed it.)

juj, 2019-09-05:
I mean instead of -DUSE_DMA_TRANSFERS=OFF, use -DUSE_DMA_TRANSFERS=ON, that will enable DMA. (also not specifying that directive at all will enable DMA, but need to delete cmakecache.txt or it will just reuse the OFF setting if it had been specified on the command line before)

The build script should detect Pi zero automatically, the ARMv6Z directive should not be manually required either.

billthecatt, 2019-09-09:
Removed the "-mfloat-abi=hard" section from CMakeLists.txt and still getting this make error:
 cmake -DCMAKE_VERBOSE_MAKEFILE=ON -DWAVESHARE_ST7735S_HAT=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DSTATISTICS=OFF -DSINGLE_CORE_BOARD=ON -DARMV6Z=ON -DUSE_DMA_TRANSFERS=ON ..
make -j
<-- snip -->
[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o -c /root/fbcp-ili9341/ssd1351.cpp
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o -c /root/fbcp-ili9341/st7735r.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/spi.cpp.o -c /root/fbcp-ili9341/spi.cpp
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o -c /root/fbcp-ili9341/statistics.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/text.cpp.o -c /root/fbcp-ili9341/text.cpp
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:79: CMakeFiles/fbcp-ili9341.dir/display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:66: CMakeFiles/fbcp-ili9341.dir/diff.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:105: CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:92: CMakeFiles/fbcp-ili9341.dir/dma.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:131: CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:170: CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:118: CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:183: CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:209: CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:157: CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:144: CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:196: CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:248: CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:222: CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o] Error 1
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
cc1plus: error: '-mfloat-abi=hard': selected processor lacks an FPU
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:261: CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:287: CMakeFiles/fbcp-ili9341.dir/text.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:274: CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:235: CMakeFiles/fbcp-ili9341.dir/spi.cpp.o] Error 1
make[2]: Leaving directory '/root/fbcp-ili9341/build'
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/fbcp-ili9341.dir/all] Error 2
make[1]: Leaving directory '/root/fbcp-ili9341/build'
make: *** [Makefile:87: all] Error 2

billthecatt, 2019-09-09:
To clarify like 49 of ../CMakeLists.txt:
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -marm -mabi=aapcs-linux -mhard-float -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations")

billthecatt, 2019-09-09:
Tried removing the: -mhard-float part.. 

I get a new error:

[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o -c /root/fbcp-ili9341/mpi3501.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o -c /root/fbcp-ili9341/mz61581.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/spi.cpp.o -c /root/fbcp-ili9341/spi.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o -c /root/fbcp-ili9341/st7735r.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o -c /root/fbcp-ili9341/ssd1351.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/text.cpp.o -c /root/fbcp-ili9341/text.cpp
/usr/bin/c++   -I/opt/vc/include  -DSINGLE_CORE_BOARD=1 -marm -mabi=aapcs-linux -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-optimizations -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=12 -DUSE_DMA_TRANSFERS=1 -DST7735S -DWAVESHARE_ST7735S_HAT -O3 -DNDEBUG   -o CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o -c /root/fbcp-ili9341/statistics.cpp
/root/fbcp-ili9341/statistics.cpp: In function 'int InitStatistics()':
/root/fbcp-ili9341/statistics.cpp:290:23: warning: no return statement in function returning non-void [-Wreturn-type]
  290 | int InitStatistics() {}
      |                       ^
In file included from /root/fbcp-ili9341/diff.cpp:6:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/display.cpp:3:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/st7735r.cpp:5:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/dma.cpp:12:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/spi.cpp:11:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/fbcp-ili9341.cpp:21:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
[... cut ...]
make[1]: Leaving directory '/root/fbcp-ili9341/build'
make: *** [Makefile:87: all] Error 2

Which I don't understand because I'm using -DSTATISTICS=OFF but I'm getting further.

Thanks again for the great project and your awesome support on all things LCD related. That said, I think the 'RPI zero' build process may need some attention. I'll happily give you whatever details I can about the environment I'm building on to help you replicate. The RPiZero and the new 1.5inch LCDs are a pretty sweet little package that I hope someday I can use.

juj, 2019-09-10:
Which OS was this on? What does ‘gcc —version’ give? Perhaps this is Pi zero on raspbian buster that updated GCC (have not tried that combo, only Raspbian Stretch on Pi zero)

billthecatt, 2019-09-10:
root@kali:~# cat /etc/issue
Kali GNU/Linux Rolling

root@kali:~# gcc --version
gcc (Debian 9.2.1-4) 9.2.1 20190821
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@kali:~# uname -a
Linux kali 4.14.80-Re4son+ #1 Fri May 17 20:25:41 UTC 2019 armv6l GNU/Linux

This is a pi zero W device running kali downloaded from here:
https://www.offensive-security.com/kali-linux-arm-images/ 

Specifically the "Kali Linux RaspberryPi Zero/Zero W" image. 

Kali is based on debian and keeps up with the stretch train so I wasn't expecting too much chaos, but then again, mebe this is an issue.

The reason I'm using this distro instead of raspbian native, is that I need the monitor grade wifi drivers in Kali to display the data on the LCD that I got it for in the first place. (I'm trying to make portable, easily deploy-able rogue AP detecting units that staff can use to find rogue APs that are in the building.)

juj, 2019-09-10:
I think that distro has a newer version of gcc than stretch has, hence the extra warnings. It can also be that it is configured differently, not sure if it might be possible for the distro to not use hard float abi at all, although why the asm code does not compile is something I have no answer to. Pi zero certainly has the rev16 instruction.

Perhaps it may be that in Kali one has to specify some other extra compiler flag to target ARMv6Z, though I don’t know what that might be.

billthecatt, 2019-09-19:
I really want this to work on the RPI zero W under kali..
 
I tried the following:

 cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DUSE_DMA_TRANSFERS=OFF -DSPI_BUS_CLOCK_DIVISOR=4 -DDISPLAY_ROTATE_180_DEGREES=ON -DSTATISTICS=0 ..

because the build was complaining about the DMA module, I tried it without... and got a valid build:

root@kali:~/fbcp-ili9341/build# make -j
[  5%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
[ 10%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
[ 15%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
[ 21%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[ 31%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
[ 36%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[ 47%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
[ 47%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o
[ 63%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/spi.cpp.o
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o
[ 89%] Building CXX object CMakeFiles/fbcp-ili9341.dir/text.cpp.o
[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o
[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
/root/fbcp-ili9341/statistics.cpp: In function 'int InitStatistics()':
/root/fbcp-ili9341/statistics.cpp:290:23: warning: no return statement in function returning non-void [-Wreturn-type]
  290 | int InitStatistics() {}
      |                       ^
In file included from /root/fbcp-ili9341/diff.cpp:6:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/display.cpp:3:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/st7735r.cpp:5:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/dma.cpp:12:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/spi.cpp:11:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
In file included from /root/fbcp-ili9341/fbcp-ili9341.cpp:21:
/root/fbcp-ili9341/spi.h: In member function 'uint32_t* SPITask::DmaSpiHeaderAddress()':
/root/fbcp-ili9341/spi.h:132:51: warning: taking address of packed member of 'SPITask' may result in an unaligned pointer value [-Waddress-of-packed-member]
  132 |   inline uint32_t *DmaSpiHeaderAddress() { return &dmaSpiHeader; }
      |                                                   ^~~~~~~~~~~~~
[100%] Linking CXX executable fbcp-ili9341
[100%] Built target fbcp-ili9341

Only to run into the following issue after running:

root@kali:~/fbcp-ili9341/build# ./fbcp-ili9341
./fbcp-ili9341: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory

Any ideas where I get/find libvchiq_arm.so? How did this even build with the library missing?

juj, 2019-09-19:
libcvhiq_arm.so is https://github.com/raspberrypi/firmware/blob/master/opt/vc/lib/libvchiq_arm.so , so perhaps it's in `/opt/vc/lib/libbvhiq_arm.so`?

billthecatt, 2019-09-19:
Sure enuf:

root@kali:~/fbcp-ili9341/build# ./fbcp-ili9341
./fbcp-ili9341: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
root@kali:~/fbcp-ili9341/build# locate libvchiq_arm.so
/opt/vc/lib/libvchiq_arm.so

Any ideas why the newly built binary can't see it?

billthecatt, 2019-09-19:
Ok, hope this helps some other poor rpi zero end user:

cd /lib/arm-linux-gnueabi
ldd /opt/vc/lib/libvchiq_arm.so
libvcos.so => not found
locate libvcos.so
ln -s /opt/vc/lib/libvcos.so .
ln -s /opt/vc/lib/libvchiq_arm.so .

~/fbcp-ili9341/build/fbcp-ili9341

Now it works! ( might be slow due to lack of DMA but this is the 1st time I've got it to work on Kali so, pretty happy!)

billthecatt, 2019-09-19:
based on the manual here:
https://www.waveshare.com/w/upload/6/6b/1.3inch_lcd_hat_user_manual_en.pdf

I'm changing the DC and reset pins as follows:

cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DUSE_DMA_TRANSFERS=OFF -DSPI_BUS_CLOCK_DIVISOR=4 -DDISPLAY_ROTATE_180_DEGREES=ON -DSTATISTICS=0 ..

and while it's very dark and very small, it actually works!

billthecatt, 2019-09-19:
Using the guide here:

https://www.raspberrypi-spy.co.uk/2014/04/how-to-change-the-command-line-font-size/ 

I cranked the font up quite a bit.. seems like I need it higher before my poor eyes can read it.. getting closer to usable.

TYVM for this awesome project!

billthecatt, 2019-09-19:
Used this resource:
https://stackoverflow.com/questions/22891235/how-to-change-screen-resolution-of-raspberry-pi

and set the following in /boot/config.txt:

hdmi_force_hotplug = 1
hdmi_cvt = 240 240 60 1 0 0 0
hdmi_group = 2
hdmi_mode = 87
overscan_left=24
overscan_right=24
overscan_top=10
overscan_bottom=24
framebuffer_width=240
framebuffer_height=240
sdtv_mode=2
sdtv_aspect=2

and now I have a working console.

billthecatt, 2019-09-19:
When I rebuild from a scratch build of kali I also had to do:

apt-get install libraspberrypi-dev

To get around several errors and missing .h files.

juj, 2020-12-08:
Closing out old issues, looks like this worked out eventually.

closed 2020-12-08 by juj