juj/fbcp-ili9341 #110: ST7789 SPI mode 3?

spinalcode, 2019-09-22:
I have a cheap ST7789 Screen that does not have the CS pin exposed. I am able to use this screen  with fbtft with the following options, I think using spi mode 3 was the main stumbling point, is it possible to enable this currently or can it be added easily?

`options fbtft_device name=flexfb gpios=reset:27,dc:25,cs:8,led:24 speed=32000000 bgr=1 fps=30 custom=1 height=240 width=240 mode=3
options flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x70,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3
`

juj, 2019-09-22:
I am not familiar with what spi mode 3 is. Tried looking at fbtft options at https://github.com/notro/fbtft/wiki/fbtft_device and spi hardware docs at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf but neither have an explanation of what spi mode 3 does. It is perhaps some fbtft internal concept?

spinalcode, 2019-09-22:
It seems to have something to do with phase timing.
http://www.rosseeld.be/DRO/PIC/SPI_Timing.htm
I can't claim to know anything about why it's important, but I could only get the screen working with mode=3 as an option. Apparently most spi devices use mode 0.

juj, 2019-09-23:
Ah gotcha. You can reverse polarity and phase e.g. via

```diff
diff --git a/st7735r.h b/st7735r.h
index 789b4b6..1a5af48 100644
--- a/st7735r.h
+++ b/st7735r.h
@@ -10,6 +10,9 @@
 // core_freq=340: CDIV=4, results in 85.00MHz, works
 // core_freq=350: CDIV=4, would result in 87.50MHz, which would work for a while, but generate random single pixel glitches every once in a few minutes
 
+// Reverse Polarity and Phase for this display
+#define DISPLAY_SPI_DRIVE_SETTINGS (BCM2835_SPI0_CS_CPOL | BCM2835_SPI0_CS_CPHA)
+
 // Data specific to the ILI9341 controller
 #define DISPLAY_SET_CURSOR_X 0x2A
 #define DISPLAY_SET_CURSOR_Y 0x2B
```

the MPI3501 display controller does that also in mpi3501.h.

spinalcode, 2019-09-23:
I added the following to st7735r.h as the screen I have is wired exactly as the https://www.waveshare.com/1.3inch-lcd-hat.htm on, without the CS pin
`
#if defined(ST7789) || defined(ST7789VW)
#define DISPLAY_NATIVE_WIDTH 240
#define DISPLAY_NATIVE_HEIGHT 240
#define DISPLAY_SPI_DRIVE_SETTINGS (BCM2835_SPI0_CS_CPOL | BCM2835_SPI0_CS_CPHA)
`
and it *almost* worked.
https://socoder.net/uploads/124/2019/39/IMG_20190923_171949.jpg

Nothing happens after the first frame is drawn, although the bottom half seems to be flickering. Any ideas what might be the problem?

hypnoshock, 2020-09-02:
Sorry to dredge this up from a year ago but did you get anywhere with this @spinalcode ?
I've got the same display and I'm getting the same result with the screen drawing about 2/3 and then bombing out on the last 1/3. I've tried compiling it with many different clock divisors such from 80 down to 20 (including 40 which is the recommended safe divisor).

The way it bombs out is like it's out of sync or something because it seems curious how the colours shift after 2/3 but the pixel output is still recognisable. I'm not sure how an SPI slave would drift out of sync if it's being fed a clock signal, can that happen? If the pixel format was wrong I wouldn't expect it to correctly draw 2/3 of the screen either. A delay on the command pin? I'm not sure how robust the timing is on that pin.

To confirm that the display wasn't broken I managed to get the display working with a python library written for the Pimroni 240x240 which also uses a st7789 at 40mhz. Out of the box it didn't work but after forcing that library to use mode 3 it worked fine. https://github.com/pimoroni/st7789-python

Nice idea with the Dreamcast VMU by the way, looks ace!

zuix14, 2020-09-06:
Hi @hypnoshock, could you share your detail to make the display works?

hypnoshock, 2020-09-06:
Hi @zuix14, I edited my last comment to make it clearer. Good luck :-)

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj