juj/fbcp-ili9341 #111: Success report: Noname 2.4" ILI9341 display

johndoe31415, 2019-09-25:
Hi there, this isn't a bug -- it's a success report. Not sure if you're interested in those?

I successfully got a noname display ("Surenoo Technologies" was the seller on Aliexpress), 2.4" with ILI9341 to work using fbcp-ili9341. The seller referenced the [documentation here](http://www.lcdwiki.com/2.4inch_RPi_Display). Built it using:

```
cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DSTATISTICS=0 ..
```

Extremely straightforward, ran immediately. Very smooth installation, extremely well-documented project, thank you so much. I'm so happy that I have a display running using code I trust instead of some arbitrary binblob :-)

Take care, Johannes

juj, 2019-09-26:
Very nice, thanks for the writeup. Makes me happy to read that the project is useful to you!

juj, 2019-09-26:
Also really cool project with BeatSaber, that game is awesome :)

closed 2019-09-26 by juj