juj/fbcp-ili9341 #116: Spi bus is just way too slow

SomeDoge, 2019-10-29:
- So, my SPI clock is extremely slow. I am getting artifacts (yellow lines) and extremely slow update rate even at 40 SPI bus clock divisor. Is it possible for soldering to be the problem?

juj, 2019-10-30:
Using SPI bus clock divisor 40 is really slow indeed. Do the artifacts persists on faster clock speeds, or does the display crash altogether?

Not sure about the artifacts and the connection to soldering. It is a bit hard to say from the video, but if the artifact pixels always appear on individual one pixel tall lines, then it is definitely not a SPI bus soldering issue, as any signal issue on the SPI bus would not be able to "know" when one line ends and another starts. I would assume if there is a soldering connectivity issue, I'd expect the whole display to just not work, or show complete garbage for a sec and then crash.

If the artifacts always persist on the same scanline, then I would be suspicious of the display unit's ribbon connection from the controller to the display panel itself, since those have parallel traces and one bad trace can make a deterministically same kind of artifact.

It is also worth checking with Waveshare's own display driver to rule out a software issue.

SomeDoge, 2019-10-30:
Okay, thanks. I will try to check the display.

SomeDoge, 2019-10-31:
So, the ribbon cable seems fine. I uploaded another video detailing what's wrong

juj, 2019-10-31:
The occasional lagging is most certainly due to low spi bus speed (sometimes the update will need to do a larger sub rectangle fill that takes much more bandwidth). If the bus does not hold up on faster settings, then you may be out of luck on that display :/

SomeDoge, 2019-10-31:
Well, that's the problem. It was working absolutely fine just a week ago on 20 SPI divisor. The ribbon cable seems fine so idk what's actually the problem.

SomeDoge, 2019-10-31:
And it's not just lagging. I don't think that the screen is supposed to refresh so that you can see the refresh going left to roght

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj