juj/fbcp-ili9341 #118: Display randomly freezes with RetroPie

lars3107, 2019-11-10:
Hello,

I have a generic ILI9341 - 2.8" 240x320 Display running with RetroPie on a Raspberry Pi 3 B+. It works very well, but it freezes randomly.
The screen is setup in portrait, so i have this line commented out in fbcp-ili9341/config.h
 #define DISPLAY_OUTPUT_LANDSCAPE

The screen was showing inverted this is why I added the following line to fbcp-ili9341/ili9341.cpp:
madctl ^= MADCTL_COLUMN_ADDRESS_ORDER_SWAP;

I also have set the resolution in /boot/config.txt to the following:

hdmi_group=2
hdmi_mode=87
hdmi_cvt=240 320 60 1 0 0 0
hdmi_force_hotplug=1

I worked my way through the FAQ, so this is what I have tried so far:
Deleted the line dtparam=spi=on in /boot/config.txt
Made sure no dtoverlay or avoid_warnings=1/2 are in /boot/config.txt
I tried altering the SPI BUS CLOCK DIVISOR. I had different even numbers from 6 up to 30, which all resulted in the screen freezing.

The game runs in the "background" but the screen only shows a frozen image. What can I do?

juj, 2019-11-10:
Hmm, try disabling DMA transfers to debug if it could be something related to that? Also there is an option to disable 9th clock cycle wait with spi transfers that is on by default - try removing that to enable the 9th clock cycle wait, in case that might help with the timings.

lars3107, 2019-11-11:
Ok so i disabled DMA transfers with:
-DUSE_DMA_TRANSFERS=OFF
Sadly this did not help. 
I was not able to find the option with the 9th clock cycle. How would i do that?

lars3107, 2019-11-11:
I solved the issue, the problem was this line in the config.h
#define DISPLAY_CONSIDERED_INACTIVE_PERCENTAGE (5.0 / 100.0)
and the fact that I was using Galaga in MAME for testing.

The problem was, in this particular game, at some point less than 5% of the screen changes, which causes the screen to turn inactive. When I tried different games everything worked perfect. Therefore when I lowered the value in config.h all games worked perfectly fine. Thank you anyways!

juj, 2020-12-08:
It looks like this turned out to be a duplicate of #102.

closed 2020-12-08 by juj