juj/fbcp-ili9341 #120: ST7735S issue

ElKoziRRo, 2019-12-05:
Hello
i`m trying to run desktop on waveshare 1.44 128x128 ST7735S but only what i can get on my display when running main loop is a weird rainbow :( i tried many compilation parameters but still only this appear what on the picture. 
Can somebody help me somehow :) ? 
[image]

juj, 2019-12-05:
In order for someone to be able to help, it would be useful to document what build flags and commands you have tried? There is not much to go with a photo of a static display otherwise.

I think I had that WaveShare hat at some point for testing, and back then it worked without much hassle. See more discussion in #29. It may be that something has regressed since if the option no longer works, so trying out the driver in an older revision may be a good if-all-else-fails fallback.

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj