juj/fbcp-ili9341 #121: problem with st7789

xinanhuang, 2019-12-07:
Hi! I got one the cheap 240x240 1.3 inch display that has the DC lines and try to emulate the pin connection to that of the waveshare display hat. so far I haven't gotten any success with both pi zero and pi 3b+. I'm on raspbian buster btw. The screen stays blank and the backlight is on the entire time. Nothing changes when I run the compiled program. I've tried both options for st7789 and st7789vw. Would really appreciate it if someone can point me towards the right direction, Thanks!

albflo, 2020-02-28:
Did you find a solution for your issue?

ClimbSnail, 2020-06-23:
You can see if it is this screen https://www.bilibili.com/video/BV1kC4y1a7fS/. I will lead out the CS pin of the screen and connect it to SPI_CE0 to work.

juj, 2020-12-08:
Closing out old issues. Feel free to reopen if still struggling with a problem.

closed 2020-12-08 by juj