juj/fbcp-ili9341 #122: UCTRONICS Display Driver?

PiAreSquared, 2020-01-04:
Hi,

I have a UCTRONICS(UC410) display that looks like its similar to the Kedei display you support.

Pin Layout:
NC  NC           MOSI  MISO  SCLK  GND
5V   5V           GND    IRQ    T_CS  L_CS

I attached a few pictures of it for better clarity. Let me know if you want to see anything else. 
[image]
[image]

I've done some digging on this display for the past few days and am trying to get it working on my Pi ZeroW. I think my display is a 3Wire SPI display(9Bit).

I can make out a few numbers on the ICs:
XPT2046 (I think this is the touch controller)
74HC595D (there are 3 of these)
AMS1117

While I have tried using the Kedei driver, it didn't seem to work. I believe it might work with the Kedei driver though because it looks very similar and the pin layout is the exact same. I'm not able to tell, but the ICs also look the exact same. If there is anything that you think might help, please let me know.

juj, 2020-12-08:
Sorry, missed this issue. Not sure what the display controller is, but if it looks identical to KeDei, then that might be it. Were you able to get this to work in the end?

gamix255, 2022-04-12:
I was also able to display the UCTRONICS 3.5 inch LCD, U4703, from the Raspberry Pi with the code of another person.
The controller is ili9488, maybe.
It is connected in parallel via the shift register.
The command starts at 0x00, 0x11. The data starts from 0x00 and 0x15.
It's very similar to KeDei's v6.3 and can actually be displayed with the code for KeDei V6.3.
The LCD seems to receive 24-bit data with the last 2 bits truncated and treat it as RGB666.

and, I tried kpishere's fork
https://github.com/kpishere/fbcp-ili9341

I was able to see it in the master, with the following options:
cmake -DSPI_BUS_CLOCK_DIVISOR=30 -DKEDEI_V63_MPI3501=ON -DBACKLIGHT_CONTROL=OFF -DSTATISTICS=0

Best regards.