juj/fbcp-ili9341 #125: Screen only works with `-DUSE_DMA_TRANSFERS=OFF`

golgobot, 2020-01-26:
## Setup
* Pi Zero W
* 1.54" 240x240 ST7789
* Stock Raspbian OS
* Compiling on the Pi

## `/boot/config.txt`

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt=240 240 60 1 0 0 0
hdmi_force_hotplug=1
```

## Build Command
When compiling with DMA transfers OFF the screen works great for about 30 seconds, then, very suddently, it suddenly turns into static.

[image]

```
cmake \
    -DST7789=ON \
    -DGPIO_TFT_RESET_PIN=24 \
    -DGPIO_TFT_DATA_CONTROL=25 \
    -DSPI_BUS_CLOCK_DIVISOR=40 \
    -DUSE_DMA_TRANSFERS=OFF \
    -DDISPLAY_ROTATE_180_DEGREES=ON \
    ..
```
When compiling with the option turned on, there's just a black screen (the backlight is on though). There are no errors from running `sudo ./fbcp-ili9341`

[image]

```
cmake \
    -DST7789=ON \
    -DGPIO_TFT_RESET_PIN=24 \
    -DGPIO_TFT_DATA_CONTROL=25 \
    -DSPI_BUS_CLOCK_DIVISOR=40 \
    -DUSE_DMA_TRANSFERS=ON \
    -DDISPLAY_ROTATE_180_DEGREES=ON \
    ..
```

I've tried playing around with RX and TX channels, but I'm not really sure what those are, so I tried different numbers for both. Always had the black screen. I've also played around with clock divisor testing numbers from 8 to 50 all with exactly the same results in both cases above.

WeiDongGitHub, 2020-03-10:
Running into the same issue on Pi Zero w https://learn.adafruit.com/2-0-inch-320-x-240-color-ips-tft-display

geogriff, 2020-10-18:
I didn't dig too far, but I noticed with the Pi Zero & Zero W, `ALL_TASKS_SHOULD_DMA` is enabled, which enables a bunch of optimizations. I got [this ST7789](https://learn.adafruit.com/adafruit-1-3-and-1-54-240-x-240-wide-angle-tft-lcd-displays) working by removing `ALL_TASKS_SHOULD_DMA` from `config.h`.

DrJaymz, 2021-04-21:
I also using pizero and nope doesn't work if ALL_TASKS_SHOULD_DMA enabled.

DrJaymz, 2021-04-21:
> I didn't dig too far, but I noticed with the Pi Zero & Zero W, `ALL_TASKS_SHOULD_DMA` is enabled, which enables a bunch of optimizations. I got [this ST7789](https://learn.adafruit.com/adafruit-1-3-and-1-54-240-x-240-wide-angle-tft-lcd-displays) working with DMA by removing `ALL_TASKS_SHOULD_DMA` from `config.h`.

That 'fixed' my problem but did it actually turn it off?