juj/fbcp-ili9341 #126: PWM brightness control

trigger7, 2020-02-02:
I've been trying to control the brightness of the backlight of my ili9341 using PWM (the LED pin is connected to a transistor for exactly that).

Using the gpio utility from WiringPi, and disabling backlight control from fbcp-ili9341, I found that some values seem to work, while others make the screen go blank.

I'm using BCM pin 13 (which has hardware PWM as far as I understand).

Values of 16, 32 and 1024 (out of 1024) seem to work ok.
64 often works, but sometimes goes blank.
128, 256, 512 don't work at all.
I haven't tried other values yet.

Is there a reason why this causes an issue ?

juj, 2020-02-02:
Does the brightness control work when not using fbcp-ili9341? That might give a clue if the driver is doing something that would be interfering with pin 13.

trigger7, 2020-02-02:
Yes, I can change the brightness of the backlight using WiringPi's gpio CLI utility :
`gpio -g mode 13 pwm
gpio -g pwm 13 <pwm_value>`

However, if <pwm_value> isn't 16, 32, 64 or 1024, as soon as run fbcp-ili9341, the screen draws what appears to be one frame, then goes blank.
If <pwm_value> is one of the above, everything works fine and the brightness is maintained.

I just tested : if I run fbcp-ili9341 and then change the PWM value, it seems to work fine !
The issue only arises when I set PWM before running fbcp-ili9341.

juj, 2020-02-02:
If you build with `-DUSE_DMA_TRANSFERS=OFF`, does that affect the issue?

juj, 2020-02-02:
According to https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf page 140, PWM DMA channel is set to channel 5, so if there is a DMA collision, it might necessitate using some other DMA channel (though [default DMA channels for fbcp-ili9341 are 7 and 1](https://github.com/juj/fbcp-ili9341/blob/master/dma.cpp#L88))

juj, 2020-02-02:
If disabling DMA does not help, GPIO modes are initialized from https://github.com/juj/fbcp-ili9341/blob/master/spi.cpp#L533, it might be interesting to probe e.g. by adding `exit(0)`s in the middle of that somewhere to gauge if there is some statement there that might cause the conflict.

trigger7, 2020-02-05:
Finally took the time to take a second look at this...

Rebuilt fbcp-ili9341 with DMA off, issue was gone. Built again with DMA on, and the issue was also gone...

I had tried  fbcp-ili9341's backlight control before (on that same pin 13), I must have forgotten to `make clean` and had leftover object files ?

Sorry for the false alarm... I will work on my project this weekend, I'll do some more testing by then.

juj, 2020-12-08:
Sorry, missed your reply. Great to hear you were able to get it to work. Maybe it was something left over from an old build.

closed 2020-12-08 by juj