juj/fbcp-ili9341 #127: Touch feature update

tousisat, 2020-02-10:
Hello,
First of all, thank you for this awesome project. It saved a lot of people from struggling with slow displays. I have actually bought the waveshare 3.5 inchs revB (https://www.waveshare.com/product/3.5inch-rpi-lcd-b.htm) and I'm wondering what is the current state of touch screen support or if you have any idea of how to make it work. 

I know this question has already been asked but there is no clear answer. Can you point me in the right direction?

Thank you in advance
Raphael Bridi

juj, 2020-02-10:
https://github.com/juj/fbcp-ili9341/pull/76 contains the most progressed development for touch support to my knowledge. You can try pulling that branch and see if it works on your WaveShare.

tousisat, 2020-02-10:
> #76 contains the most progressed development for touch support to my knowledge. You can try pulling that branch and see if it works on your WaveShare.

I really appreciate your quick reply. I still have problems building the pull request. I asked the question on #76 

Thank you again!

closed 2020-12-08 by juj