juj/fbcp-ili9341 #128: Compatibility with other SPI devices

OopsDoh, 2020-02-10:
Is it possible to use other SPI devices in combination with fbcp-ili9341 installed?

I have a **Raspberry Pi Zero W** with an [2.0" 320x240 Display](https://www.adafruit.com/product/4311) (ST7789) and I want to add [this 24-Channel LED driver board with SPI](https://www.adafruit.com/product/1429) for additional LED indicators.

As far as I understand the [RasPi Pinnout shown here](https://de.pinout.xyz/pinout/spi#) it should be possible to use BCM**19** (SPI1 MISO), BCM**16** (SPI1 CE2), BCM**20** (SPI1 MOSI) and BCM**21** (SPI1 SCLK) for the LED board - instead of the pins that are blocked by the display.

In the [description for the LED driver](https://learn.adafruit.com/tlc5947-tlc59711-pwm-led-driver-breakout/python-circuitpython) they explain to use the Adafruit Blinka lib. This requires activating SPI output as explained [here](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi). They also metion to add 

> dtoverlay=spi1-3cs

to the bottom of **/boot/config.txt** so there can be used multiple SPI devices.

So... is it possible to use the "second" SPI for addional sensors or boards like the LED driver?

In your install instructions you mention that the default Raspi SPI should be disabled to avoid compatibility problems. Is there a way to use both, the display with fbcp-ili9341 AND another SPI devices?

juj, 2020-02-10:
Short answer is that I don't know.

Longer version, theoretically it should be possible to utilize SPI1 and SPI2 (and the newer SPIx on a Pi 4), although care must be taken that the software driving the other SPI buses does not stomp on fbcp-ili9341, and that fbcp-ili9341 does not stomp on the other driver.

"Theoretically", because it is hard to evaluate how well this is working in practice, until one goes down that route and does the actual development. It is likely that such compatibility work may need software modifications to both fbcp-ili9341 and the other driver to cooperate.

OopsDoh, 2020-04-09:
Thanks for your answer. Unfortunately changing the SPI interface is too complex for me to distribute here to the fbcp-ili9341 project.

[But I finally found a way to extend Adafruits PiTFT installer script (with fbtft) to support the 2.0" 320x240 display.](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/80) HDMI mirroring and video playback on the TFT screen is running quite smoothly and I can use the second SPI for the led board in parallel without problems. I think the display even runs with 60fps, but didn't test it with gaming applications.

I still want to thank you: The code in your st7735r.cpp file helped me to better understand how the init sequence for the display is working.

juj, 2020-04-10:
Great to hear that you found a setup that works! Good luck with the build!

closed 2020-12-08 by juj