juj/fbcp-ili9341 #130: Pi Zero + Waveshare 3.5 v.2 ILI9486 (IPS) - Wrong colors

daviel, 2020-03-21:
Hello,

I am using this Display (https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)) with the raspberry pi zero. But whatever build parameters I use the colors seem to be off. See screenshot.

[image]

I used the following build parameters:

cmake .. -DWAVESHARE35B_ILI9486=ON -DUSE_DMA_TRANSFERS=OFF -DSPI_BUS_CLOCK_DIVISOR=40 -DDISPLAY_INVERT_COLORS=ON

Here without inverted colors:

[image]

I recognized that -DDISPLAY_SWAP_BGR=ON does not even have a effect on the colors. 

What am I doing wrong? I am using the shield pin layout. Using the waveshare default waveshare drivers it is working with the correct colors. Thank you for any advice. :)

juj, 2020-03-21:
Hmm.. hard to say from that screenshot.

Perhaps you can try viewing some color test images on the display, e.g.

[image]
[image]
[image]

My hunch is that the display is not in RGB565 mode, but is perhaps in RGB555 mode instead. This does not seem like a BGR<->RGB swap since the red "-4" text comes out correctly (that would show up as blue if BGR and RGB were swapped).

A random guess might be to change all occurrences of `VC_IMAGE_RGB565` (found in gpu.cpp) to `VC_IMAGE_TF_RGBA5551`.

daviel, 2020-03-21:
Thank you for the fast response. Exchanging the VC_IMAGE_RGB565 entries with VC_IMAGE_TF_RGBA5551 does not work. fbcp keeps crashing then with this error message:

```
vc_dispmanx_snapshot() failed with return code -1! 
If this appears related to a change in HDMI/display resolution, see https://github.com/juj/fbcp-ili9341/issues/28 and https://github.com/raspberrypi/userland/issues/461 (try setting fbcp-ili9341 up as an infinitely restarting system service to recover)
Quit.
```

Here are screenshots taken with the testimages above. But I don't know how to interpret them. 

[image]
[image]
[image]

daviel, 2020-03-21:
Taking a look at the fbcp which is provided by waveshare it shows that they are using VC_IMAGE_RGB565 too. See here: https://github.com/waveshare/LCD-show/blob/286df2fc7aadd01340b138bda5fef3b327508119/rpi-fbcp/main.c#L56. Is this relevant?

juj, 2020-03-21:
> fbcp keeps crashing then with this error message:

Oh too bad.. :/ That means that DispmanX does not support capturing the framebuffer in RGB555 format. That might have been a quick way to fix.

The results are really weird indeed.. The first image suggests to me that perhaps the display would be running in RGB444 mode, i.e. 4 bits per color channel, but the banding is smooth and not hard cut, so that makes me doubt if that is the right mode either.

In any case, to make it work, there's two ways:

1) In case the display does have a RGB565 mode, it should be set with this command: https://github.com/juj/fbcp-ili9341/blob/master/ili9486.cpp#L41 . According to ILI946 data sheet, that command should accept either RGB565 mode, or RGB666 mode (https://datasheetspdf.com/pdf-file/945603/ILITEK/ILI9486/1), but there is no RGB444 mode. It is possible that the display is not a ILI9486.

2) If the display cannot be identified and the data sheet found, it will be necessary to do a software conversion to the display format that the display initializes with. Coding a bit blind, but this patch might work to convert from RGB565 to RGB444 if that might be the format:
```diff
diff --git a/fbcp-ili9341.cpp b/fbcp-ili9341.cpp
index 88d242d..86aedfa 100644
--- a/fbcp-ili9341.cpp
+++ b/fbcp-ili9341.cpp
@@ -502,6 +502,21 @@ int main()
           data = (uint16_t*)((uintptr_t)data + 3);
         }
 #else
+        // Convert from R5G6B5 to R4G4B4
+        const int numRedBits = 4;
+        const int numGreenBits = 4;
+        const int numBlueBits = 4;
+        while(x < endX)
+        {
+          uint16_t pixel = scanline[x];
+          uint16_t r = pixel >> 11;
+          uint16_t g = (pixel >> 6) & 0x3F;
+          uint16_t b = pixel & 0x1F;
+          r >>= 5 - numRedBits; // From 5 bits to 4 bits.
+          g >>= 6 - numGreenBits; // From 6 bits to 4 bits.
+          b >>= 5 - numBlueBits; // From 5 bits to 4 bits.
+          scanline[x++] = (r << (numGreenBits + numBlueBits)) | (g << numBlueBits) | b;
+        }
         while(x < endX && (x&1)) *data++ = __builtin_bswap16(scanline[x++]);
         while(x < (endX&~1U))
         {
```
You could try that patch and see how it works. Definitely some kind of conversion should succeed. Setting e.g. `g = 0; b = 0;` instead of `g >>= 6 - ...` and `b >>= 5 - ...` might help solve one color channel at a time.

juj, 2020-03-21:
Another idea might be to generate a test pattern with
```diff
diff --git a/fbcp-ili9341.cpp b/fbcp-ili9341.cpp
index 88d242d..4dfe207 100644
--- a/fbcp-ili9341.cpp
+++ b/fbcp-ili9341.cpp
@@ -502,6 +502,11 @@ int main()
           data = (uint16_t*)((uintptr_t)data + 3);
         }
 #else
+        int x2 = x;
+        while(x2 < endX)
+        {
+          scanline[x2++] = 1 << y;
+        }
         while(x < endX && (x&1)) *data++ = __builtin_bswap16(scanline[x++]);
         while(x < (endX&~1U))
         {
```
That should output horizontal stripes throughout the display, that would "solve" what color channels each bit in the output correspond to.

daviel, 2020-03-22:
The change from above gives me this result:
[image]

The other change this:
[image]

daviel, 2020-03-22:
I have requested additonal documentation from waveshare. I hope we can find out more through this. They have linked a pdf here: https://www.waveshare.com/w/upload/2/2f/3.5inch-rpi-lcd-b-panel-dimension.pdf which says that the driver for this display is an ILI9486L which seems quite weird.

juj, 2020-03-22:
Hmm, the test pattern came out a bit small.. Can you help decode the second screenshot a little closer?

Counting from the top to bottom, are there 6 scanlines each for blue, green and then red? E.g.
```
darkest blue
slightly less dark blue
less dark blue
moderate blue
brighter blue
brightest blue
```
Then same for green, and then for red?

Perhaps this code might generate a clearer pattern with thicker lines:

```
diff --git a/fbcp-ili9341.cpp b/fbcp-ili9341.cpp
index 88d242d..4dfe207 100644
--- a/fbcp-ili9341.cpp
+++ b/fbcp-ili9341.cpp
@@ -502,6 +502,11 @@ int main()
           data = (uint16_t*)((uintptr_t)data + 3);
         }
 #else
+        int x2 = x;
+        while(x2 < endX)
+        {
+          scanline[x2++] = (y % 8 < 4) ? 0 : (1 << (y/8));
+        }
         while(x < endX && (x&1)) *data++ = __builtin_bswap16(scanline[x++]);
         while(x < (endX&~1U))
         {
```
Fbcp-ili9341 does have a dedicated driver backend for ILI9486L, you could try that out as well.

daviel, 2020-03-22:
Unfortunately the changed scanline-code does not show any lines. 

Also using the build parameters for the ili9486L does not show anything on the screen...

juj, 2020-03-23:
Hmm, that is odd... If changing controllers on the fly to CMake line, that may actually be due to file CMakeCache.txt getting stale - try deleting that (or the whole `build/` subdirectory) for a fresh rebuild?

daviel, 2020-03-23:
I have written a small script which deletes the build directory and rebuilds fbcp everytime in order to prevent any caching.

daviel, 2020-03-23:
Okay, forget about what I said. I removed the -DUSE_DMA_TRANSFERS=OFF in my build script for performance testing. Now it works as expected.

daviel, 2020-03-23:
> > fbcp keeps crashing then with this error message:
> 
> Oh too bad.. :/ That means that DispmanX does not support capturing the framebuffer in RGB555 format. That might have been a quick way to fix.
> 
> The results are really weird indeed.. The first image suggests to me that perhaps the display would be running in RGB444 mode, i.e. 4 bits per color channel, but the banding is smooth and not hard cut, so that makes me doubt if that is the right mode either.
> 
> In any case, to make it work, there's two ways:
> 
>     1. In case the display does have a RGB565 mode, it should be set with this command: https://github.com/juj/fbcp-ili9341/blob/master/ili9486.cpp#L41 . According to ILI946 data sheet, that command should accept either RGB565 mode, or RGB666 mode (https://datasheetspdf.com/pdf-file/945603/ILITEK/ILI9486/1), but there is no RGB444 mode. It is possible that the display is not a ILI9486.
> 
>     2. If the display cannot be identified and the data sheet found, it will be necessary to do a software conversion to the display format that the display initializes with. Coding a bit blind, but this patch might work to convert from RGB565 to RGB444 if that might be the format:
> 
> 
> ```diff
> diff --git a/fbcp-ili9341.cpp b/fbcp-ili9341.cpp
> index 88d242d..86aedfa 100644
> --- a/fbcp-ili9341.cpp
> +++ b/fbcp-ili9341.cpp
> @@ -502,6 +502,21 @@ int main()
>            data = (uint16_t*)((uintptr_t)data + 3);
>          }
>  #else
> +        // Convert from R5G6B5 to R4G4B4
> +        const int numRedBits = 4;
> +        const int numGreenBits = 4;
> +        const int numBlueBits = 4;
> +        int x2 = x;
> +        while(x2 < endX)
> +        {
> +          uint16_t pixel = scanline[x2];
> +          uint16_t r = pixel >> 11;
> +          uint16_t g = (pixel >> 6) & 0x3F;
> +          uint16_t b = pixel & 0x1F;
> +          r >>= 5 - numRedBits; // From 5 bits to 4 bits.
> +          g >>= 6 - numGreenBits; // From 6 bits to 4 bits.
> +          b >>= 5 - numBlueBits; // From 5 bits to 4 bits.
> +          scanline[x2++] = (r << (numGreenBits + numBlueBits)) | (g << numBlueBits) | b;
> +        }
>          while(x < endX && (x&1)) *data++ = __builtin_bswap16(scanline[x++]);
>          while(x < (endX&~1U))
>          {
> ```
> 
> You could try that patch and see how it works. Definitely some kind of conversion should succeed. Setting e.g. `g = 0; b = 0;` instead of `g >>= 6 - ...` and `b >>= 5 - ...` might help solve one color channel at a time.

Setting DISPLAY_COLOR_FORMAT_R6X2G6X2B6X2 as a define in the ili9486.h does repair the white color but it totally damages the image. Also rerendering is broken then...

juj, 2020-03-23:
Yeah, the display is definitely not in DISPLAY_COLOR_FORMAT_R6X2G6X2B6X2 mode, because that mode uses 24 bits to specify one pixel. In the current mode it has been passing 16 bits per pixel, and since the pixel geometry lines up properly there, the display is definitely in 16bpp mode and not in 24bpp mode.

daviel, 2020-03-23:
> Yeah, the display is definitely not in DISPLAY_COLOR_FORMAT_R6X2G6X2B6X2 mode, because that mode uses 24 bits to specify one pixel. In the current mode it has been passing 16 bits per pixel, and [...]

Do you have any idea left what I could also try? Would be really sad if this display is not supported as it has a really good image quality :(

juj, 2020-03-23:
> This picture shows that there are 5 bits for blue and red and 6 bits for green? The conversion should then work right?

Oh hmm, that is really odd. Indeed, the image proves that there are 5 bits for blue, 6 for green and 5 for red, just like the code is expecting in the first place.

Btw, is it just an odd effect of the camera, but it looks to me as if on each color channel, the second to last row is the brightest? E.g. for blue:
```
very dim blue
dim blue
moderate blue
very bright blue
not as bright blue as before
```
Same for green and red? It looks like second to last row are the brightest? It would be expected that the intensity would grow line by line, last row being the most intense/brightest on each color channel?

daviel, 2020-03-23:
> Oh hmm, that is really odd. Indeed, the image proves that there are 5 bits for blue, 6 for green and 5 for red, just like the code is expecting in the first place.
>
> Btw, is it just an odd effect of [...]

You're right. The second last line is the brightest one of them.

juj, 2020-03-23:
Hmm, that is really odd. It seems as if the panel is RGB565, but then the bits inside each color are somewhat scrambled.

Perhaps the following kind of test pattern would be useful to gauge what is happening:

[image]

My guess is that either there is some oddity causing a scrambled bit pattern, or then there is some initialization with respect to some voltage levels or refresh rates something like that, that is missing, which causes the odd color palette.

Perhaps we should be looking at the following test images:

[image]

[image]

First one comes with banding lines, the second one should be a smooth gradient. How do those look like? I'm thinking perhaps looking at the output of those images might reveal a way to unscramble the bits.

juj, 2020-03-23:
Btw, is the visual output "stable"? I.e. there is no brightness or voltage fluctuation, and no flickering?

daviel, 2020-03-23:
> Hmm, that is really odd. It seems as if the panel is RGB565, but then the bits inside each color are somewhat scrambled.
> 
> My guess is that either there is some oddity causing a scrambled bit pattern, or then there is some initialization with respect to some voltage levels or refresh rates something like that, that is missing, which causes the odd color palette.
> 
> Perhaps we should be looking at the following test images:
> 
> [image]
> 
> [image]
> 
> First one comes with banding lines, the second one should be a smooth gradient. How do those look like? I'm thinking perhaps looking at the output of those images might reveal a way to unscramble the bits.

There is many information lost due to the camera of my smartphone but in the gradient picture it is clearly visible that the colors get dark about 25% from top. Same for the other picture with banding lines.

[image]
[image]

daviel, 2020-03-23:
> Btw, is the visual output "stable"? I.e. there is no brightness or voltage fluctuation, and no flickering?

No the image is very stable I can't recognize any flickering or somehting. 

I have been playing around with the waveshare driver and could overclock the SPI drive up to 60Mhz which showed some flickering but the colors did not become changed as with fbcp.

juj, 2020-03-23:
That is very peculiar indeed.

Oh, I wonder if this could be connected to gamma control: in ili9486.cpp, there are lines

```c
    SPI_TRANSFER(0xE0/*Positive Gamma Control*/, 0x00, 0x2C, 0x2C, 0x0B, 0x0C, 0x04, 0x4C, 0x64, 0x36, 0x03, 0x0E, 0x01, 0x10, 0x01, 0x00);
    SPI_TRANSFER(0xE1/*Negative Gamma Control*/, 0x0F, 0x37, 0x37, 0x0C, 0x0F, 0x05, 0x50, 0x32, 0x36, 0x04, 0x0B, 0x00, 0x19, 0x14, 0x0F);
```
Try commenting those out. (That will also need a hard power off of the display)

Reading the spec at https://www.displayfuture.com/Display/datasheet/controller/ILI9486L.pdf, there is a paragraph about "Z inversion". That feature should not be enabled - it should be already disabled by default at power-off, but you could try explicitly adding a command
```c
    SPI_TRANSFER(0xB4, 0);
```

Another try may be to comment out some other initialization directives in the file, in case they might have an effect here.

daviel, 2020-03-23:
> That is very peculiar indeed.
>
> Oh, I wonder if this could be connected to gamma control: in ili9486.cpp, there are lines
>
> ```c
>     SPI_TRANSFER(0xE0/*Positive Gamma Control*/, 0x00, 0x2C, 0x2C, [...]

Wow, you did it!! Commenting out all the gamma controls fixes the colors! Do you have paypal or anything? Get a beer as a thank you on my cost. :)

[image]
[image]
[image]

closed 2020-03-24 by juj

juj, 2020-03-24:
Wow, nice! Great to hear that it worked out! Very interesting to know that the gamma control fields can affect the display this dramatically.

Thanks for the beer offer also, much appreciated. Although I don't have a Paypal account.. :)

daviel, 2020-03-24:
> Wow, nice! Great to hear that it worked out! Very interesting to know that the gamma control fields can affect the display this dramatically.
> 
> Thanks for the beer offer also, much appreciated. Although I don't have a Paypal account.. :)

Is there any way I can support you and your work? :)

juj, 2020-04-03:
Thanks again and sorry for the delay in replying. I might see about adding a PayPal donate button on the repository at some point. For now, it would be cool if you ever get around to making a project showcase build video, blog or screenshots, those are always enjoyable to read! :)

PR #131 (2020-03-24): added skip gamma control flag for waveshare 35b (v2)
fixes #130 

I hope it is okay to introduce this flag like this.