juj/fbcp-ili9341 #139: Geekcreit screen all white

xmaayy, 2020-04-02:
# Description

[This it the screen](https://www.banggood.com/Geekcreit-3_5-inch-TFT-LCD-Touch-Screen-Protective-Case-Touch-Pen-Kit-For-Raspberry-Pi-3B3B2B-p-1391232.html?rmmds=category&cur_warehouse=CN)

The board has no real markings for its maker but this is their pin definition
[image]
Seems like it would be similar to #40. They also use the same driver download site as the other 3 pin screen issue `git clone https://github.com/goodtft/LCD-show.git`. 

# Steps taken
Tried using `-DKEDEI_V63_MPI3501=ON` and `-DSPI_BUS_CLOCK_DIVISOR=30 (all the way to 100)` and it sometimes flickered (guessing that was the CLS) but nothing after that. Always a white screen.

juj, 2020-04-03:
Very hard to say, I'm afraid. Nothing stands out on the web page.

If trying out the different build drivers and options does not help, it may be that the only option is to reverse engineer the bus.

xmaayy, 2020-04-03:
Darn. The odd part is I install the drivers and the touchscreen works (I can move the mouse around with it), but it just stays white. So I can tell its communicating with at least one part of the screen properly.

Think its a dud?

juj, 2020-04-03:
Do you mean that the display stays white even with the goodtft LCD-show drivers? (since fbcp-ili9341 does not have touch support, except in https://github.com/juj/fbcp-ili9341/pull/76 branch) If so, then the display may be faulty, try contacting manufacturer for support?

RafhaanShah, 2020-04-04:
I have the same one (http://www.lcdwiki.com/3.5inch_RPi_Display) works fine with goodtft drivers but just getting a white screen with fbcp-ili9341. Any ideas?

briwa, 2020-05-06:
First off, thank you for this project @juj. I was following [this guide](https://retropie.org.uk/forum/topic/4157/guide-retropie-3-5-inch-rpi-lcd-v3-0-no-brand) for my 3.5 inch screen and was not satisfied with the FPS and yours gave me hope and it was definitely an improvement. Thank you again.

From that link you'll probably see that we might have a similar looking TFT screen (I got mine from [this link](https://my.cytron.io/p-3.5-inch-tft-touch-screen-for-rpi-3b-3b-plus-4b) ([wiki](http://www.lcdwiki.com/3.5inch_RPi_Display) that they provide). I got mine working using: 
```
-DWAVESHARE35B_ILI9486=ON
```

because at first both simply looked pretty similar (although in retrospect it wasn't exactly the case). Turns out, it works!

I also used:
```
 -DSPI_BUS_CLOCK_DIVISOR=16
```
Anything below that wasn't really working for me. I was thinking if I should go to 20 but seems like 16 works fine.

Here's a little demo:
[image]

Heavier games like PSX sure drops but at this point I'm already satisfied with it, since it's better than the alternative.

Not sure if it applies to any of your situatio. I'm just excited that I managed to get it working and just hoped that this could help anyone. Cheers

juj, 2020-05-06:
That's pretty cool! Something that I found on the ILI9486 that I tested was that since it is so much more bus constrained compared to ILI9341, it created tricky framerate "bounces" when in static scenes it would work 60fps, but when there is a lot of action, framerate would quickly tank. Something I found interesting for that was to set [TARGET_FRAME_RATE](https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/display.h#L6) to 30 instead of 60.

That would reduce the overall frame rate, but in sudden action scenes, frame rate fluctuations would be smoother, so in some games it was a bit more pleasing visually. Might be worth a shot..

briwa, 2020-05-07:
@juj I tried your suggestion, and while it's not producing perfect 60fps, it does help not having that sudden bounces! Thank you again, I really appreciate your work.

closed 2020-12-08 by juj

zeekoe, 2022-02-03:
For future reference, I also have this display and needed to disable DMA for it to display something:

cmake -DSPI_BUS_CLOCK_DIVISOR=30 -DWAVESHARE35B_ILI9486=ON -DUSE_DMA_TRANSFERS=OFF ..