juj/fbcp-ili9341 #141: fbcp-ili9341 not running at startup

NorvillesDingus, 2020-04-09:
I have set everything up but the screen wont work when I power everything on. The back light turns on but no display. I can start it with terminal, but since I want to use the screen as the main display, that's not going to work.
I'm super new to everything and this is my first project, so I'm guessing I'm just doing something wrong.

The settings used are: -DSPI_BUS_CLOCK_DIVISOR=6 -DWAVESHARE_ST7789VW_HAT=ON

juj, 2020-04-10:
It is hard to say what might be wrong here. Try following the different items in the troubleshooting section, if one of those might help. Also using the display's official driver first is recommended to make sure that the display hardware is working properly.

NorvillesDingus, 2020-04-11:
I found my solution. I just wasn't putting the code into the right spot. This is where it ended up for me. I just didn't understand where to put it.

fi
sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &
exit 0

closed 2020-12-08 by juj

MichalkapDeveloper, 2022-02-25:
Add this` sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &` to /etc/rc.local.
EXAMPLAE:
`#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

sleep 7
fbcp &

con2fbmap 1 0

sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &

exit 0
`