juj/fbcp-ili9341 #142: Native spi display not read correctly

owenreynolds43, 2020-04-27:
Hi, worked great on a 2.4” 320 x 240 screen using ili9341 driver. Needed to change to a 2” 320 x 240 ST7789V screen for size restrictions. It displays the screen but detects It as a 240 x 240 screen so left half is missing. Is there a simple fix?

juj, 2020-04-28:
You can try editing https://github.com/juj/fbcp-ili9341/blob/d6fafd67bdd46876d1a8a3e53e3bd50544cc13e4/st7735r.h#L19 to set up 320x240.

If the display scan order is portrait, then it will require setting up height=320 instead of width.

owenreynolds43, 2020-04-28:
Wow. Thankyou for your quick reply, I’ll get on to it straight away👍👍

owenreynolds43, 2020-05-02:
Sorry if this not the place to ask but I cannot get the auto start to work with rc.local.
During boot it starts then when the boot goes to a black screen it doesn’t stay loaded.
I have a 1st gen model B running Rasbian Buster.
Everything works if I start from terminal.
Thanks for the other help regards sizing👍
Regards
Owen

juj, 2020-05-02:
Hmm, hard to say. that is not an issue I have run into before. I can't remember if I ever tried autostart on Raspberry Pi 1.

Try double checking that the Raspbian SPI driver is disabled from /boot/config.txt.

Also it may be possible that the display changes resolution between boot and desktop. That is known to crash DispmanX. See also https://github.com/juj/fbcp-ili9341/issues/28 , https://github.com/juj/fbcp-ili9341/issues/96 and https://github.com/juj/fbcp-ili9341/issues/118 , those may relate to the issue.

owenreynolds43, 2020-05-02:
Fixed it👍 Had to add sudo command
sudo /home/pi/fbco-ili9341/build/fbcp-ili9342 in 
etc/xdg/lxsession/LXDE-pi/autostart 
above  @xssreensaver.
Permission set accordingly.
Thanks anyway appreciate the good work you have done.
Regards
Owen

closed 2020-12-08 by juj