juj/fbcp-ili9341 #146: no-brand 2.0" 240x320 display with ST7789V

Darmur, 2020-06-16:
I recently got a no-brand 2.0" screen from Aliexpress
https://www.aliexpress.com/item/4000832671813.html

[image]

[image]

[image]

[image]

[image]

According to the description and to the sticker on the box, it is based on ST7789V with resolution 240x320

I'd like to have it working with RPi, size and resolution is great for a portable RetroPie setup.

This is the connection I applied:

```
GND -> RPi Pin 9
VCC -> RPi Pin 17
SCL -> RPi Pin 23 (GPIO11, SPI_CLK)
SDA -> RPi Pin 19 (GPIO10, SPI_MOSI)
RST -> RPi Pin 13 (GPIO27)
DC -> RPi Pin 22 (GPIO25)
CS -> RPi Pin 24 (GPIO8, SPI_CE0)
```

There is no pin for blacklight control

This is the command I'm using for selecting build options:

This is the cmake command I am using for selecting build options:

`cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=40 -DUSE_DMA_TRANSFERS=OFF ..`

I tried to build leaving DISPLAY_NATIVE_WIDTH 240 and DISPLAY_NATIVE_HEIGHT 240 in st7735r.h, but also trying to change both to 320 (not at the same time, of course)

Result is always the same: black screen.

Am i doing something wrong?

juj, 2020-06-16:
Nothing in particular stands out here. Worth testing with the provided driver that it works well. If it does, it may be a case to debug via sniffing the bus for differences.

Darmur, 2020-06-16:
thanks for quick response. I'll ask the seller some documentation, at the moment I don't have any driver working

juj, 2020-12-08:
Closing old issues.

closed 2020-12-08 by juj

Darmur, 2021-01-02:
thanks to the tricks from a youtube video, I could finally have the display up and running with your library.

I had to edit those two files:

st7735r.h
line 20 -> #define DISPLAY_NATIVE_HEIGHT **320**

st7735r.cpp
line 95 -> SPI_TRANSFER(0x37/*VSCSAD: Vertical Scroll Start Address of RAM*/, 0, **0**);

it would be nice to add a build option for supporting the ST7789 displays with full resolution (320x240)

lewilou22, 2023-06-06:
Thank you so much, it took me 2 dyas to find this fix @Darmur