juj/fbcp-ili9341 #147: SEPS525 supported?

toto99303, 2020-06-17:
Hello,

Is it possible to add support for SEPS525 display?
I'm using this display in a VMU mod with Pi Zero, but the refresh rate is killing the experience...

https://sudomod.com/forum/viewtopic.php?t=6917

I hope something could be done about this...

juj, 2020-06-17:
The display does look like it could be supported within current driver design. I don't unfortunately have the display (or even if I did, unfortunately not able to spend time to work on adding support at the moment).

If someone has the display and is able to find a spec sheet or reverse engineer the bus, pull requests would be welcome!

toto99303, 2020-06-17:
As far as I read the bus is very well described in the attached datasheet?
Unfortunately my programming skills are not enough to pull this off, that's why I asked here...
Since the bus i documented I could try, but I would need some guidelines how to start.

juj, 2020-06-17:
Oh right - indeed, the spec sheet documents the protocol. In general the porting process is as follows:
1) Write the initialization sequence for the display. E.g. ILI9341 has it [here](https://github.com/juj/fbcp-ili9341/blob/master/ili9341.cpp#L30)
2) Write the pixel cursor update sequence protocol, e.g. for ILI9341 [here](https://github.com/juj/fbcp-ili9341/blob/master/ili9341.h#L27).
3) Write any needed display-specific color conversion e.g. [here](https://github.com/juj/fbcp-ili9341/blob/master/fbcp-ili9341.cpp#L491)
4) Address any other display-specific aspects that the display is incompatible with against the existing driver (e.g. partial window sets, DMA, or something else)

toto99303, 2020-06-17:
Thanks for the fast answers! I'll try to find time to dig into this.