juj/fbcp-ili9341 #149: ST7789V 240*240 without CS PIN

ClimbSnail, 2020-06-23:
Do I have to connect the CS pin? I used ST7789V 240*240 screen and found that CE did not work without connection. The PCB of my screen module does not have CS pin, how can I make the screen work without CE?

juj, 2020-06-23:
If a display does not have a chip select, it is then intended to work without one. Fbcp-ili9341 does not need to connect CS/CE pin for the driver’s sake.

ClimbSnail, 2020-07-02:
But I tried, and the screen couldn't show. It is said that ST7789 works in mode 3 instead of mode 0 by default when there is no CS. Is this right? Without CS, I verified in 51 and arduino that the screen is normal, and the code of 51 uses the simulated SPI of Mode 3.

juj, 2020-07-02:
SPI mode is specified in code by the `#define DISPLAY_SPI_DRIVE_SETTINGS` field.

E.g. MPI3501 display driver changes to mode 3 by having this line: https://github.com/juj/fbcp-ili9341/blob/d2f69320fb165d2ec95fa33ea4d06d865c49f56e/mpi3501.h#L32

You could try replacing this line https://github.com/juj/fbcp-ili9341/blob/52484caa6998aa779539a03d69ec346984b1109a/display.h#L83 with the above from mpi3501.h.

juj, 2020-07-02:
> It is said that ST7789 works in mode 3 instead of mode 0 by default when there is no CS. Is this right?

In general whether a display uses a Chip Select pin or not is orthogonal to the Polarity and Phase settings of the communication on the Data & Clock pins. But it is possible that the specific ST7789 display you have would be configured in this way.

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj