juj/fbcp-ili9341 #150: Is here support for GamePi20?

petrmachcz, 2020-07-02:
Hello, I do not understand hardware, numbering chips etc. unfortunately. Is it this driver compatible with GamePi20? I using it with standard Raspbian Pi OS Lite. Thank you for the advice.

juj, 2020-07-02:
Unfortunately there is no developed support for GamePi20. Looking at https://www.waveshare.com/gamepi20.htm and https://www.waveshare.com/wiki/GamePi20, WaveShare does not seem to publicize the hardware specifications of the display controller, so it is difficult to know if the setup would be utilizing one of the display controllers that fbcp-ili9341 already supports.

Btw, in the past I have attempted to contact WaveShare directly regarding some other products they publish, to ask for a data sheet. Their reply has been dismissive, implying they'd only consider sharing such information to customers buying in bulk (thousands/tens of thousands units?)

Although there does exist a flexfb-based fbtft.conf script in https://www.waveshare.com/wiki/GamePi20. That could be used as a basis to develop at least partial driver support for the device, as it identifies the pin numberings and one suitable initialization sequence. It could be likely that the display is a good old ILI9341, which does work quite well.

petrmachcz, 2020-07-02:
Thank You for your reply.

Jpe230, 2021-02-20:
Sorry for reviving this issue:

My Gamepi20 arrived a week ago and I was about to test fbcp-ili9341 with the display, sadly while assembly the "game console" I shattered the display rendering it useless.

It seems that the display used in this console is very similar to these one: [https://www.waveshare.com/wiki/2inch_LCD_Module](https://www.waveshare.com/wiki/2inch_LCD_Module) and [https://www.adafruit.com/product/4311](https://www.adafruit.com/product/4311). The LCD Ribbon cable looks very similar, so maybe the controller is a ST7789 although I couldn’t test it.

Looking by the init code given by the wiki and looking at [st7735r.cpp](https://github.com/juj/fbcp-ili9341/blob/master/st7735r.cpp) it differs a bit but mostly it seems to match.

Once my new gamepi20 arrives I will start making more tests.