juj/fbcp-ili9341 #151: Display JPG?

WillMo2198, 2020-07-07:
Hi,

I have an ili9341 TFT LCD wired up to my RPi. I have used this driver to show the output of the screen itself (command line since I'm using Raspbian Lite), but I just want to display a single JPG image. Is there any way to use this driver to do that, (maybe with fbi or something) or should I just use another driver?

Thanks

juj, 2020-07-07:
The driver is just a driver. It looks like the [fbi](https://www.raspberrypi-spy.co.uk/2017/02/how-to-display-images-on-raspbian-command-line-with-fbi/) program might do the trick, or you can also look for some other driver as well if there is one that has support built-in.

WillMo2198, 2020-07-07:
thanks! I got it working by `sudo fbcp-ili9341/build/fbcp-ili9341 | fbi -d /dev/fb0 -T 1 /path/to/img.jpg`

closed 2020-07-07 by WillMo2198