juj/fbcp-ili9341 #153: Washed Out Unreadable 3.2" ILI9341 (Especially Dark Text On White Background)

TheFlav, 2020-07-10:
I've noticed an issue where certain text is almost unreadable on the 3.2" ILI9341.  This seems to be something of a contrast issue, but I'm no expert.

I decided to test some things, and it turns out that the Positive and Negative Gamma Correction values have a definite effect on this.  Again, I'm no expert, so I have no idea what these values should be or what the specific bytes are controlling.  

What I did was to look around at some other drivers.  I found ILI9341 init code at https://www.displaytech-us.com/forum/ili9341-initialization-code that was doing the gamma differently.  So, I hacked https://github.com/juj/fbcp-ili9341/blob/master/ili9341.cpp and commented out these two lines 

=====
//    SPI_TRANSFER(0xE0/*Positive Gamma Correction*/, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E, 0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00);
//    SPI_TRANSFER(0xE1/*Negative Gamma Correction*/, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31, 0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F);
=====

and added these two lines with the values from the displaytech link above.
=====
    SPI_TRANSFER(0xE0/*Positive Gamma Correction*/, 0x1f, 0x1a, 0x18, 0x0a, 0x0f, 0x06, 0x45, 0x87, 0x32, 0x0a, 0x07, 0x02, 0x07, 0x05, 0x00);
    SPI_TRANSFER(0xE1/*Negative Gamma Correction*/, 0x00, 0x25, 0x27, 0x05, 0x10, 0x09, 0x3a, 0x78, 0x4d, 0x05, 0x18, 0x0d, 0x38, 0x3a, 0x1f);
=====

This made the aforementioned text readable.  I tried to do some A/B test to see which version I liked better for some other screens, and I'd say I like the original way better for some things, but the edited lines make certain text readable.  I'll opt for readable over unreadable if it's at the expense of a slight preference elsewhere.

I also tried this configuration (only changing the Negative Gamma):
=====
    SPI_TRANSFER(0xE0/*Positive Gamma Correction*/, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E, 0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00);
//    SPI_TRANSFER(0xE1/*Negative Gamma Correction*/, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31, 0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F);

//    SPI_TRANSFER(0xE0/*Positive Gamma Correction*/, 0x1f, 0x1a, 0x18, 0x0a, 0x0f, 0x06, 0x45, 0x87, 0x32, 0x0a, 0x07, 0x02, 0x07, 0x05, 0x00);
    SPI_TRANSFER(0xE1/*Negative Gamma Correction*/, 0x00, 0x25, 0x27, 0x05, 0x10, 0x09, 0x3a, 0x78, 0x4d, 0x05, 0x18, 0x0d, 0x38, 0x3a, 0x1f);
=====

That resulted in readable text, too.  

So, I'm not sure where to go with this.  I don't know what the values are changing, exactly, but I do know that readable is better than unreadable.  

Any thoughts?

juj, 2020-07-11:
I also have no idea about the exact meaning of the gamma control directives, they are not documented anywhere that I’d have seen. There are also some voltage divide control settings that have a similar effect.

Hard to say what would be best for everyone. Perhaps some comparing photos before and after might help people decide which way they would like to go. Also not sure if different ILI9341 would prefer different settings (it is possible this is very pabel specific)

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj

MisutaaUrufu, 2021-04-15:
@juj Correct me if I'm wrong, but isn't it setting the gamma curve for each color? I'm having trouble parsing it, but http://www.lcdwiki.com/res/MSP2807/ILI9341%20Datasheet.pdf contains a section on the "**GAMSET**" command (_P107_), which is what the initialization code from ilitek uses to set the "**gamma curve**", and sections on the "**PGAMCTRL**" (_0xE0_) (_P188_) and "**NGAMCTRL**" (_0xE1_) (P189) commands, which are what the actual values being written by fbcp-ili9341.

It describes them as being used to "**Set the gray scale voltage to adjust the gamma characteristics of the TFT panel.**". I'm not quite clear on how to parse the documentation however, so I have little understanding of the actual values being set. But i think it appears to be adjusting the voltage in grayscale of each color (RGB) to effect its saturation/brightness? — Perhaps you'd have a better time parsing it than I.

For what it's worth, _the values from ilitek, mentioned by @TheFlav,_ work better for my waveshare 3.2b clone; the colors were washed out in applications with the default values.

juj, 2021-04-16:
> @juj Correct me if I'm wrong, but isn't it setting the gamma curve for each color?

Yes, that is what they do, but I don't know how to interpret the values to resolve issues like this. The existing gamma curve values are copy-paste from documented initialization sequence pdfs.

tuupola, 2021-04-16:
Gamma curves etc are panel specific. They do not have anything to do with drive chip (ILI9341) itself. It is kind of pointless a display driver to include them unless you know what panel you are going to use. Most drivers just copy them from some other code without realising this.

TheFlav, 2021-04-16:
> Gamma curves etc are panel specific. They are not related to the driver chip (in this case the ILI9341) itself. It is kind of pointless a display driver to include them unless you know what panel you are going to use. Most drivers just copy them from some other code without realising this.

This is interesting, tuupola.  I hadn't considered this.  We have received 3.2" ILI9341 panels from multiple suppliers.  We have never found an IPS panel, but we have found some with different glass and viewing angle configurations.  It now makes more sense to me why some could look better/worse than others and some might improve (or get worse) just by trying other gamma settings.  For us, it would be really nice if we could have some way of selecting gamma settings at runtime, so it could be tested just by changing a parameter and rebooting (or restarting the driver).