juj/fbcp-ili9341 #154: Turn off display before shutdown?

zarcade, 2020-07-31:
Is it possible to turn off this display as part of my python shutdown script? It stays white at the moment.

juj, 2020-08-01:
If the display turns white, it means that it is still being powered from its voltage line, but the display controller has been turned off. To turn off the display light, you will have to stop feeding it power; or, if the display has a controllable backlight pin, wire that in and use that to stop the backlight.

One middle-ground workaround may be to not turn the display off at all, so the display would stay on, showing a black screen, up until all the power is cut off. You can try looking up the function `void TurnDisplayOff()` for your display in fbcp-ili9341 codebase, and edit that out so the display is not deinitialized at all - then it should keep showing a black screen.

zarcade, 2020-08-02:
Thanks. I simply added a line to the shutdown script to pull the led gpio low. Did the trick.

rigues, 2020-10-14:
> Thanks. I simply added a line to the shutdown script to pull the led gpio low. Did the trick.

Can you share how did you do that? I'm facing the same issue.

juj, 2020-12-08:
Closing out old issues.

It sounds like @zarcade used something like http://wiringpi.com/the-gpio-utility/ to manipulate the GPIO state at shutdown. E.g. `gpio write <pin> 0`.

closed 2020-12-08 by juj