juj/fbcp-ili9341 #155: Waveshare 3.5 A border issue

T0biasCZe, 2020-08-05:
I, have setuped Waveshare 3.5 model A, its almost same as model B you have tested, except it have TN instead of IPS and it doesnt have inverted colours.
cmake line i used is `cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=14`
I set core freq to 400 and CDIV to 14, i compiled it and it show picture without any problem, but the screen is bordered, it shouldnt be bordered, because the screen is 480x320, and i set screen resolution to 960x640 which is perfectly 2x as large.
It looks like this:
[image]
is it be fixable ?

juj, 2020-08-05:
That does look odd, I cannot immediately think of a reason that would be causing those borders. What happens if you were to make the display size in /boot/config.txt to 480x320 to match the size 1:1. Does it still cause the borders?

Do the garbage pixels on the border area change occassionally, or is the garbage the same static pixels after boot?  It looks like some of that garbage even bleeds over to the nonbordered area?

T0biasCZe, 2020-08-05:
Ok, i fixed the problem with borders, it was because i set the screen resolution with "framebuffer_width=x" and "framebuffer_height=x". The hdmi was outputed in that resolution, but FBCP saw it as Full HD.
But the garbaging still occurs when lot of pixels is updated (for example when i left click on desktop and i pull)
is this normal or not ? 
Video: https://drive.google.com/file/d/1E4MpB0QMwNT3jkSML7umzZfAsBE67Fki/view?usp=sharing

juj, 2020-08-05:
That error looks some kind of desynchronization with display update window commands, so the updates end up being performed in wrong parts of the screen. Basically the pixel data is going in sync, but command data is getting desynchronized. Some things that come to mind:

- add `#define DISPLAY_WRITE_PIXELS_CMD_DOES_NOT_RESET_WRITE_CURSOR` to ili9486.h and rebuild.
- try disabling DMA from CMake line and rebuild
- replace `#define UNLOCK_FAST_8_CLOCKS_SPI() (spi->dlen = 2)` in spi.cpp:102 with `#define UNLOCK_FAST_8_CLOCKS_SPI() (spi->dlen = 0)` and rebuild.
- try a higher CDIV setting

You can try them all in combination to start with, they should all be compatible changes. If none of that helps, another possibility is that the protocol command codes for update window specification is different on that display.

T0biasCZe, 2020-08-06:
#define DISPLAY_WRITE_PIXELS_CMD_DOES_NOT_RESET_WRITE_CURSOR didnt work
#define UNLOCK_FAST_8_CLOCKS_SPI() (spi->dlen = 0) causes error when compiling (make -j, adding log)
[output.txt](https://github.com/juj/fbcp-ili9341/files/5036464/output.txt)

also, would it be possible to add optional feature for rewriting whole screen each X frames or seconds ?
for example -REWRITE_WHOLE_FRAMES=30 would rewrite it each 30th frame and -REWRITE_WHOLE_SECOND=3 each third frame
It would be useful in non game projects when you dont need high refresh rate but need quality (and maybe games that dont need high refresh rates?)
because the garbage happens on places where the screen isnt updated at all, places which are updated lot dont have it

juj, 2020-08-07:
> #define UNLOCK_FAST_8_CLOCKS_SPI() (spi->dlen = 0) causes error when compiling (make -j, adding log)
> [output.txt](https://github.com/juj/fbcp-ili9341/files/5036464/output.txt)

It looks like a typo (missing '#' in the beginning of the line?)

> also, would it be possible to add optional feature for rewriting whole screen each X frames or seconds ?
> for example -REWRITE_WHOLE_FRAMES=30 would rewrite it each 30th frame and [...]

To change the refresh rate, edit display.h

https://github.com/juj/fbcp-ili9341/blob/52484caa6998aa779539a03d69ec346984b1109a/display.h#L5-L6

(see the documentation at https://github.com/juj/fbcp-ili9341#tuning-cpu-usage )

T0biasCZe, 2020-08-07:
> To change the refresh rate, edit display.h

no, you dont get what i mean
i mean that it could rewrite whole screen each X seconds instead of only pixels that updated
it could fix the issue with garbage
i think its happening when pixel change, the software capture it, and the pixel change so fast from the original that the software dont capture it (my theory), so then there are ghost pixels until something overwrites it
it would be good optional settings for something like this

juj, 2020-08-07:
To write the whole screen, instead of doing delta updates, enable the option `#define UPDATE_FRAMES_WITHOUT_DIFFING` in

https://github.com/juj/fbcp-ili9341/blob/52484caa6998aa779539a03d69ec346984b1109a/config.h#L108-L111

(this is also documented in the readme that was already [linked in above](https://github.com/juj/fbcp-ili9341#tuning-cpu-usage))

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj