juj/fbcp-ili9341 #156: LG4573A LCD Display likely compatible?

CuriousJames, 2020-08-08:
Hello,

I just found this repo and I'm hoping it will support the LCD Touch-Screen display I found on AliExpress (I know that site doesn't have a great rep, but I've found them to be pretty good).

This is the item: https://www.aliexpress.com/item/32573391663.html?spm=a2g0o.cart.0.0.6a283c00RpVae4&mp=1

It's got the LG4573A driver for the LCD and an MXT224 for the touch screen.

I saw it can run on SPI, so I'm hopeful it could work, but although I've tinkered with many things with a Raspberry Pi, a touch screen is not one of them so I'm rather new to this, and would be very glad of any pointers.   Ideally I'm looking for a fairly high res capacitive touch screen that's around 3" that's under £20, but if my only options are more expensive then I'll concede on that front.

Thanks to anyone that's willing to help me!

juj, 2020-08-08:
Hi, unfortunately fbcp-ili9341 does not support touch screen. There is a fork at https://github.com/kpishere/fbcp-ili9341/ that adds some amount of touch support on some controller (iirc the controller name started with "XPT").

Also fbcp-ili9341 does not support LG4573A, this is actually the first time I come across that display controller, that is not familiar from before.

juj, 2020-08-08:
Looking at the link, that does not seem like an SPI-based display. It is perhaps a DPI device? No datasheet available on that page.

Generally recommend getting one of the tested displays listed on the main page, others will likely not work.

CuriousJames, 2020-08-08:
I've managed to get the datasheet if that helps at all?
[LG4573A_DS_V1.1.5.pdf](https://github.com/juj/fbcp-ili9341/files/5045214/LG4573A_DS_V1.1.5.pdf)

Is there any display you could point me to that would work with the fork of this repo that also supports touch screen by any chance?  I know that's a big ask so no problems if you can't.

James.

juj, 2020-08-08:
The datasheet confirms it is a DPI -based display.

Try asking in the fork repository about which touch devices and displays it supports?

CuriousJames, 2020-09-07:
Thanks for your advice!

closed 2020-09-07 by CuriousJames