juj/fbcp-ili9341 #157: Set up freeze on raspberry pi

dh928, 2020-08-11:
it gets stuck at running main loop now(maybe it works but takes a long time?). I am trying to set up the driver on my raspberry pi4.
[image]
what should I do?

juj, 2020-08-11:
At that stage the driver is running normally. If nothing is showing up on the screen, then it means that something is going wrong with the messaging to the display. Check through the troubleshooting checklist on the main readme page for ideas what to try.

dh928, 2020-08-12:
> At that stage the driver is running normally. If nothing is showing up on the screen, then it means that something is going wrong with the messaging to the display. Check through the troubleshooting checklist on the main readme page for ideas what to try.

Thank you! I change bus clock I visor to 30 and it worked!

closed 2020-08-12 by dh928