juj/fbcp-ili9341 #158: How can I measure how fast my display is updating?

dh928, 2020-08-12:
I have a SSD1351 OLED display running on raspberry pi4, I ran my program with very low resolution and got FPS as high as 500+, but the display can't actually update that fast, it's not possible at 20 MHz SPI. I wonder if it's possible to see how fast the display is actually updating. I know there we can get the SPI bus speed=core_freq/CDIV, but that is not what I’m looking for. I’m looking for how much data is required to refresh a 128x128 16bit screen and figure out the full screen refresh rate based on that. Does anyone know how I can do it?

juj, 2020-08-12:
When you build with statistics enabled, it will show on the top left what the actual fps update rate at any given time is.

To figure out what the full screen update rate is, set the pi up to display some animated content. If the fps text is show in red instead of white, then it means the spi bus cannot keep up with the content.

dh928, 2020-08-12:
> When you build with statistics enabled, it will show on the top left what the actual fps update rate at any given time is.
>
> To figure out what the full screen update rate is, set the pi up to [...]

Thank you so much!!!!

dh928, 2020-08-12:
> When you build with statistics enabled, it will show on the top left what the actual fps update rate at any given time is.
>
> To figure out what the full screen update rate is, set the pi up to [...]

there were some numbers with P after it, like 55P or something on my display after I turned the statistics on. I am not sure what that means in terms of fps

juj, 2020-08-12:
Check out the video https://www.youtube.com/watch?v=dqOLIHOjLq4 for details on the different fields.

closed 2020-12-08 by juj