juj/fbcp-ili9341 #159: aarch64 partial success / waveshare 2in ST7789VW

atisharma, 2020-08-24:
I was able to compile on 64b raspberry pi official-unofficial image.

    $ uname -a
    Linux pluto 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

I had to edit CMakeLists.txt to use
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mlittle-endian -funsafe-math-optimizations")`
and to recognise my board revision `b03112`.

I have one of these:

https://www.waveshare.com/2inch-lcd-module.htm
https://www.waveshare.com/wiki/2inch_LCD_Module

and it runs, but only with some partial success.

```
bcm_host_get_peripheral_address: 0xfe000000, bcm_host_get_peripheral_size: 25165824, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 500000000hz, max turbo: 500000000hz. SPI CDIV: 40, SPI max frequency: 12500000hz
Initializing display
Resetting display at reset GPIO pin 27
Creating SPI task thread
InitSPI done
Bus error
```
The display clears (well, the first 240x240 LHS of it, which is to be expected).

using
`cmake -DSPI_BUS_CLOCK_DIVISOR=40 -DST7789VW=ON -DGPIO_TFT_RESET_PIN=27 -DGPIO_TFT_DATA_CONTROL=25 -DUSE_DMA_TRANSFERS=OFF ..`

Also tried with DMA on, and get the same Bus error.

boot.config has
```
framebuffer_width=240
framebuffer_height=240
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt=240 240 60 1 0 0 0
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
arm_64bit=1
disable_splash=1
hdmi_blanking=2
hdmi_force_hotplug=1
```

Is this due to 64 bit #43 / #79  or a similar problem to issue #146 / #121?

juj, 2020-12-08:
Sorry, missed this issue.

I am new to this `Bus error` message, that is not something that I remember that fbcp-ili9341 would print out. Were you able to resolve the issue?

atisharma, 2020-12-11:
I gave up and used a different screen, sorry.

leo60228, 2021-10-31:
> I am new to this `Bus error` message, that is not something that I remember that fbcp-ili9341 would print out.

"Bus error" means that SIGBUS was delivered to the process (like "Segmentation fault" for SIGSEGV). I think this is the signal that would be delivered when trying to access an invalid address in /dev/mem, which makes snese.