juj/fbcp-ili9341 #160: ST7796S Support

furrysalamander, 2020-09-09:
I just purchased a 4" 480*320 ST7796S Based Display off AliExpress ([link](https://www.aliexpress.com/item/4000497625786.html)).  I'm hoping to use it in a project my club is building.  It doesn't look like the board has shift registers on it, so I imagine that's probably a good sign for getting high SPI clock speeds.  

I found the datasheet for this display controller here:
https://www.displayfuture.com/Display/datasheet/controller/ST7796s.pdf

I'm hoping that it will be fairly similar to the ST7789 or the ST7735 so that a port will be fairly easy.

furrysalamander, 2020-09-09:
This page here seems to indicate that this controller will work quite well:
http://www.lcdwiki.com/MHS-4.0inch_Display-B

They list a maximum SPI speed of 125MHz.  I don't know for sure if that will apply to the display I have purchased, but it leaves me optimistic.  They also have examples for how to get this display working on the Raspberry Pi, so that's another good sign.

furrysalamander, 2020-09-22:
I got the screen yesterday.  I was hoping I'd at least get a good head start from the info on the page linked to above:

http://www.lcdwiki.com/MHS-4.0inch_Display-B

But no such luck.  I'm still optimistic that it will be a good starting point for me, though perhaps the display controller datasheet will prove more useful in the long run.

I tried using the pre-built images they provide on the page, but I wasn't able to get anything besides a white screen, and sometimes when I restarted, I'd manage to get a bit of noise on the screen.  Aside from that though, no progress.  I'm still learning how SPI displays work, but I started digging into config.txt, and it looks like it gets setup on this line:

```dtoverlay=mhs395```

Digging around yields the device tree files, `mhs395.dtbo` and `mhs395-overlay.dtb`.  I'll see if I can get any more useful information from them.

One thought that I had is that maybe the screen I got can't handle the extremely high SPI bus speed that the pre-built image is configured to (125MHz).  They're getting 50FPS without any extra tricks.  I'm not sure where I'd go to turn down the bus speed, but hopefully I can figure it out.

furrysalamander, 2020-09-22:
I was able to decompile the dtb and dtbo files using dtc
`dtc -I dtb -O dts mhs395-overlay.dtb > mhs395-overlay.dts`

[mhs395.txt](https://github.com/juj/fbcp-ili9341/files/5263382/mhs395.txt)
[mhs395-overlay.txt](https://github.com/juj/fbcp-ili9341/files/5263383/mhs395-overlay.txt)

furrysalamander, 2020-10-01:
Great news!  I've basically got it working.  I made a fork where I'm keeping track of my changes.  

Right now everything seems to be working, except the display is mirrored, so everything is displayed backwards.  Hopefully I can figure that out soon.  

[image]

furrysalamander, 2020-10-01:
I was able to fix this problem in my latest commit.  If I use the following cmake options, everything displays correctly:

```
cmake -DST7796S -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DGPIO_TFT_BACKLIGHT=24 \
-DSPI_BUS_CLOCK_DIVISOR=4 -DDISPLAY_SWAP_BGR=ON -DBACKLIGHT_CONTROL=ON ..
```

I'm actually quite pleasantly surprised.  The display seems to run just fine at 100MHz (400MHz/4), and could maybe be driven even faster (I haven't played around with the divisor settings too much.  

However, I'm still having a few more issues.  I'm getting all kinds of ghosting and mild tearing when things are in motion.  I suspect that the Pi is perhaps handing over data faster than the controller is putting it out onto the LCD?  Honestly, it's hard to say.  @juj perhaps you might have a better idea as to what's causing the problems?  The pi seems to be pushing out 60fps without any major hiccups.  I'm including some pictures of this behavior:

[image]

[image]

As you can see in the following photo, if the game is paused, the display looks flawless otherwise:

[image]

I really hope we can figure this out, since this is a very affordable display, and it's 4" and 480x320!  It would be perfect for so many portal raspberry pi projects.

furrysalamander, 2020-10-01:
I was able to fix the tearing issues by using the frame rate values from the device tree files linked above.
IE, command B1 (FRMCTR1)

All my changes are on the latest commit on the fork I made.  Just tested with a clean build on my Pi and I was able to use the display without any issues.

Here's the options I used for CMAKE with the latest commit:

```
cmake -DST7796S=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DGPIO_TFT_BACKLIGHT=24 -DSPI_BUS_CLOCK_DIVISOR=4 -DBACKLIGHT_CONTROL=ON ..
```

Once my pull request gets approved, this issue should be resolved.

juj, 2020-12-08:
Thanks for working on that support. I went ahead and ordered the same display from AliExpress, it will take some weeks to arrive. It does look like an impressive 480x320 display controller, probably on the level of the Tontec one that is no longer in sale.

Electronlibre2012, 2021-08-07:
hello @furrysalamander and @juj 

great work!

i have buy this screen and i thought it was a ILI9488 but it doesnt work so i read the item description on Aliexpress and its a ST7796S : [this](https://fr.aliexpress.com/item/1005002632269797.html?spm=a2g0s.9042311.0.0.27426c372EtgnY)

i finally make some search and find your post...so start from the beginning, delete the folder and clone your repo instead of ju ju's one.

I have a white screen, that doesnt miror the HDMI screen.

i have this warning : Performance warning: using slow memcpy_to_dma_and_prev_framebuffer_in_c() function. Check conditions in display.h that enable OFFLOAD_PIXEL_COPY_TO_DMA_CPP and configure to use that instead.

I have make it works for a ILI9341 3.5" and 2.8" without issue and follow this post [](https://krystof.io/mirroring-raspberry-pi-hdmi-video-to-a-st7789-1-3-inch-lcd-display/) where the autor say to add this line `madctl ^= MADCTL_COLUMN_ADDRESS_ORDER_SWAP;` before `SPI_TRANSFER(0x36/MADCTL: Memory Access Control/, madctl);` .

Sorry for my poor knowledge...is there some more settings to do please?

I dont find a ST7796S.h or .cpp ...dont understand.

Please can you help me a little please...so frustrating to have this screen blank...

Thanks

okilonet1, 2024-05-07:
I tried setting up as you did, with a rpi 3b+ and thesame display, 

but keep getting:
wrapper@raspberrypi:~ $ sudo ./fbcp-ili9341/build/fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 4, SPI max frequency: 100000000hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75cc3000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75cc3700, DMA RX channel register file is at ptr: 0x75cc3100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 27
Setting TFT backlight on at pin 24
Creating SPI task thread
InitSPI done
vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt

when I go to /boot/config.txt I get redirected to /boot/firmware/config.txt

Eques12, 2024-08-31:
> Great news! I've basically got it working. I made a fork where I'm keeping track of my changes.
> 
> Right now everything seems to be working, except the display is mirrored, so everything is displayed backwards. Hopefully I can figure that out soon.
> 
> [image]

Can you please tell me how to do this? I have the same display practically trying to do the same thing but none of the display library's im using are working