juj/fbcp-ili9341 #163: ST7735S  black screen

Merith-TK, 2020-09-15:
Built with 
```
pi@merithpi:~/Workspace/fbcp-ili9341/build $ cmake -DWAVESHARE_ST7735S_HAT=ON  -DSPI_BUS_CLOCK_DIVISOR=6 ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Doing a Release build
-- Board revision: a02082
-- Detected this Pi to be one of: Pi 2B rev. 1.2, 3B, 3B+, 3A+, CM3, CM3 lite or 4B(1GB,2GB,4GB RAM), with 4 hardware cores and ARMv8-A instruction set CPU.
-- Enabling optimization flags that target ARMv8-A instruction set (Pi 2B >= rev. 1.2, 3B, 3B+, CM3 or CM3 lite)
-- Scaling source image to view. If the HDMI resolution does not match the SPI display resolution, this will produce blurriness. Match the HDMI display resolution with the SPI resolution in /boot/config.txt to get crisp pixel perfect rendering, or alternatively pass -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON to crop instead of scale if you want to view the center of the screen pixel perfect when HDMI and SPI resolutions do not match.
-- Preserving aspect ratio when scaling source image to the SPI display, introducing letterboxing/pillarboxing if HDMI and SPI aspect ratios are different (Pass -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON to stretch HDMI to cover full screen if you do not care about aspect ratio)
-- SPI_BUS_CLOCK_DIVISOR set to 6. Try setting this to a higher value (must be an even number) if this causes problems. Display update speed = core_freq/divisor. (on Pi3B, by default core_freq=400). A safe starting default value may be -DSPI_BUS_CLOCK_DIVISOR=40
-- USE_DMA_TRANSFERS enabled, this improves performance. Try running CMake with -DUSE_DMA_TRANSFERS=OFF it this causes problems, or try adjusting the DMA channels to use with -DDMA_TX_CHANNEL=<num> -DDMA_RX_CHANNEL=<num>.
-- Targeting WaveShare 128x128 1.44inch LCD Hat with ST7735S controller
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Workspace/fbcp-ili9341/build
pi@merithpi:~/Workspace/fbcp-ili9341/build $ make -j
Scanning dependencies of target fbcp-ili9341
[  4%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
[  9%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
[ 14%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[ 19%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
[ 23%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
[ 28%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
[ 33%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
[ 38%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[ 42%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9488.cpp.o
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
[ 57%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o
[ 61%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o
[ 66%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
[ 71%] Building CXX object CMakeFiles/fbcp-ili9341.dir/low_battery.cpp.o
[ 76%] Building CXX object CMakeFiles/fbcp-ili9341.dir/spi.cpp.o
[ 80%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
[ 85%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o
[ 90%] Building CXX object CMakeFiles/fbcp-ili9341.dir/text.cpp.o
[ 95%] Building CXX object CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o
[100%] Linking CXX executable fbcp-ili9341
[100%] Built target fbcp-ili9341
pi@merithpi:~/Workspace/fbcp-ili9341/build $ sudo ./fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75b7f000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75b7f700, DMA RX channel register file is at ptr: 0x75b7f100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 27
Creating SPI task thread
InitSPI done
DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
Relevant source display area size with overscan cropped away: 480x720.
Source GPU display is 480x720. Output SPI display is 130x129 with a drawable area of 128x128. Applying scaling factor horiz=0.18x & vert=0.18x, xOffset: 23, yOffset: 1, scaledWidth: 85, scaledHeight: 128
Creating dispmanX resource of size 85x128 (aspect ratio=0.664062).
GPU grab rectangle is offset x=0,y=0, size w=85xh=128, aspect ratio=0.664062
All initialized, now running main loop...
```

 Pi3b+

Merith-TK, 2020-09-15:
```
# /boot/config.txt
hdmi_force_hotplug = 1
hdmi_cvt = 128 128 60 1 0 0 0
hdmi_group = 2
hdmi_mode = 87

# /etc/rc.local ## the command is where i store my built binaries
/opt/userbin/fbcp &
```

Merith-TK, 2020-09-15:
built with #77 and it works... although display is squished to hell

juj, 2020-09-20:
Take note of the message
```
Source GPU display is 480x720.
```
in the log. If you have 

```
hdmi_cvt = 128 128 60 1 0 0 0
```

then fbcp-ili9341 should be saying that the source GPU display is 128x128. Given that it is not, it means that some piece of software has resized the HDMI framebuffer size from 128x128 to 480x720. Try to remove running such software, or try to configure that to also adapt to 128x128 resolution.

You can also try using the `tvservice` and `fbset` command line tools to adjust the resolution after boot, but that should be done while `fbcp-ili9341` is running, because changing the resolution will crash Pi's DispmanX library (issue https://github.com/raspberrypi/userland/issues/461 )

Merith-TK, 2020-09-20:
Thats part of the problem. nothing in my system would controll the display at all, and surprisingly, it doesnt crash, the resize works

I have fbcp running as a systemd service to clean things up and make things easier (litterally, just running fbcp)
Using the Python SPI library for this it works without issue and scales properly.

However no matter what i do, if it the x and y res values are the same, i get a boarder that shrinks my display space by atleast 10-20px on all sides.

Merith-TK, 2020-09-20:
And the only way i am able to get the display to work at a readable resolution, is by using fbset

currently,

I have my Pi setup with tty auto login
`fbset 128x128-60`
The fbmode i have setup so fbset even works, Otherwise i get a larger display signal than what i want
```
mode "128x128-60"
    # D: 25.175 MHz, H: 31.469 kHz, V: 59.94 Hz
    geometry 128 128 128 128 32
    timings 39722 48 16 33 10 96 2
endmode
```

`/boot/config.txt`
```
# For more options and information see
# http://rpf.io/configtxt

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=-10
overscan_right=0
overscan_top=0
overscan_bottom=0

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=87
hdmi_cvt = 128 128 60 1 0 0 0
ctv_width=130
ctv_height=129

cvt_framerate=60
cvt_aspect=1
cvt_margins=0
cvt_interlace=0
cvt_rb=1

# uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=spi=on

dtparam=audio=on

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1

## GPIO DISPLAY ##
display_rotate = 2
```

Merith-TK, 2020-09-20:
and even when i set `framebuffer_width/height` to 128, i still get this nasty boarder

juj, 2020-09-20:
What does `tvservice -s` give?

https://www.raspberrypi.org/forums/viewtopic.php?t=25933 suggests one can run 

```
tvservice -m CEA
tvservice -m DMT
```
to list the available modes. There should be a custom mode string to set arbitrary resolution like 128x128.

I don't know if /boot/config.txt is sensitive to spaces, so might try having `hdmi_cvt=128 128 60 1 0 0 0` instead of `hdmi_cvt = 128 128 60 1 0 0 0`.

Also try removing `dtparam=spi=on`, that can interfere with fbcp-ili9341.

The options
```
cvt_framerate=60
cvt_aspect=1
cvt_margins=0
cvt_interlace=0
cvt_rb=1
overscan_left=-10
overscan_right=0
overscan_top=0
overscan_bottom=0
```
should not be needed, try removing those as well. (fbcp-ili9341 can also be configured to do cropping)

Merith-TK, 2020-09-20:
```
pi@merithpi:~ $ tvservice -s
state 0xa [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive
pi@merithpi:~ $ tvservice -m CEA
Group CEA has 2 modes:
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
           mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
pi@merithpi:~ $ tvservice -m DMT
Group DMT has 1 modes:
  (prefer) mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive
```

Merith-TK, 2020-09-20:
your suggested changes fixed the boarder issue (aside from a small but tolerable amount on the bottom)

juj, 2020-09-21:
Tvservice should be showing the 128x128 mode. This suggests something has reset it, or the /boot/config.txt field did not apply.

Check out e.g. 
https://www.raspberrypi.org/forums/viewtopic.php?t=24679 for more discussion about custom modesetting.

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj