juj/fbcp-ili9341 #164: Startup

lolax80, 2020-09-15:
Hello,

how can I see the startup seriell log on the screen?
Is there an option to hide the status bar from your driver?

Tahnks

juj, 2020-09-20:
Sorry, I do not understand what you mean. What do you mean by the startup seriell log?

lolax80, 2020-09-20:
I wanted to view the boot console on the tft screen

juj, 2020-09-20:
Since the display driver is implemented in software, and not into the GPU firmware, the display can only boot up to show contents when the fbcp-ili9341 software is run. /etc/rc.local runs probably somewhat late to the boot sequence. https://www.raspberrypi.org/forums/viewtopic.php?t=31567 discusses setting up system services, which might boot up earlier.

Also maybe messing with the linux runlevels, something like https://www.liquidweb.com/kb/linux-runlevels-explained/ (/etc/rc0.d or /etc/rc1.d), might give a chance to hook into earlier in the bootup sequence. Although not sure, I am not very familiar with that mechanism.

lolax80, 2020-09-20:
Ok, thanks

closed 2020-12-08 by juj