juj/fbcp-ili9341 #165: make -j error

Brentlok, 2020-09-19:
When I try to run "make -j" command everything goes wrong, here's log : 
Scanning dependencies of target fbcp-ili9341
[  4%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[  9%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
[ 14%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9488.cpp.o
[ 19%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
[ 23%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
[ 28%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
[ 33%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
[ 42%] Building CXX object CMakeFiles/fbcp-ili9341.dir/low_battery.cpp.o
[ 42%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
[ 47%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
[ 57%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
[ 61%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
[ 71%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o
[ 71%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:102: CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o] Błąd 1
make[2]: *** Oczekiwanie na niezakończone zadania....
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:141: CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o] Błąd 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:63: CMakeFiles/fbcp-ili9341.dir/diff.cpp.o] Błąd 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:76: CMakeFiles/fbcp-ili9341.dir/display.cpp.o] Błąd 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:115: CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o] Błąd 1
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:128: CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o] Błąd 1
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ms sysv
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:180: CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o] Błąd 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:167: CMakeFiles/fbcp-ili9341.dir/ili9488.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:154: CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:89: CMakeFiles/fbcp-ili9341.dir/dma.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:206: CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:219: CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-mlittle-endian’
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:245: CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o] Błąd 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:193: CMakeFiles/fbcp-ili9341.dir/low_battery.cpp.o] Błąd 1
c++: error: unrecognized command line option ‘-mlittle-endian’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:232: CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o] Błąd 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/fbcp-ili9341.dir/all] Błąd 2
make: *** [Makefile:84: all] Błąd 2

I'm using the newest retropie

juj, 2020-09-20:
That looks odd. Can you try running the following commands on the command line, and paste their output?

```
cat /etc/*-release
lsb_release -a
uname -a
cat /proc/version
c++ -v
```

Is this on Raspberry Pi 4?
My guess is that Retropie images are based on some other distro configuration than Raspbian, or has updated the compiler to something that is no longer compatible with the build flags.

You can also try deleting or commenting out this line:

https://github.com/juj/fbcp-ili9341/blob/9766d43dce9e406404b76ba76dc71daa0d4c3b1b/CMakeLists.txt#L49

Brentlok, 2020-09-20:
I have tried to use older version of retropie (3.8) but its still not working, deleting 49 line in CMakeLists.txt did nothing too. Something is wrong but I dont know what. Month ago I didn't have any issues. Maybe you could just send me a compiled file with parameters -DSPI_BUS_CLOCK_DIVISOR=6 -DADAFRUIT_ILI9341_PITFT=ON, btw thanks for reply

juj, 2020-09-20:
Oh hmm, when deleting the cmake line, one may need to also delete the file CMakeCache.txt in the build directory (or do a build in a new directory) so that it will not pick up the directive from stale cache.

Brentlok, 2020-09-20:
I dont know how but after reupload whole repo I could build app without any issue. Thanks for your help anyways!!!

closed 2020-12-08 by juj