juj/fbcp-ili9341 #166: Ili9341 will not initialise if playing intro video with retropie

zarcade, 2020-09-20:
Just noticed that after getting the display to work, I decided to add a loading video to retropi as it takes a while for the display to initialise and adding some kind of sound would let the user know it was booting. However adding a splash video stops the display initialising. I confirmed by removing the video and the display initialises again.

Anyone else get this?

juj, 2020-09-21:
There was an earlier report about this kind of behavior, apparently a video player stomped over the DMA channel that fbcp-ili9341 was using.

Another possibility is that the video player might be resizing the display size on the fly, that dows not work well with other applications actively utilizing DispmanX.

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj