juj/fbcp-ili9341 #170: SPI stops talking to ST7789

krzysztofkuczek, 2020-10-01:
Hi,
I'm trying to compile it for little 240x240 ST7789 display (1,3inch from https://pl.aliexpress.com/item/32950879136.html) and it does not work.
Trying to investigate what's is going on i found out that after less then 1s from reset SPI stops sending any data.
What do i wrong?
[image]

Kris
PS
I was trying with waveshare 128x128 display ant it works fine, so rpi hardware (rpi 3 v 1.2) seems to be ok.

juj, 2020-10-09:
That is hard to say.. certainly nothing is moving there.

Does the same hang happen with both DMA enabled and disabled?

juj, 2020-12-08:
Closing out old issues.

closed 2020-12-08 by juj