juj/fbcp-ili9341 #173: aarch64 Arch Linux - Bus error (core dumped)

gearhead, 2020-10-23:
Trying this out on a B3+ with a pitft3.5 from adafruit. I run Arch Linux and am running kernel 5.4.70 with the RPi build config (dtoverlays work, etc)
Read through the issues and instructions and think I did what was needed and made the binary this way:
edited the CMakeLists.txt to comment out a line to change the build flags:
```
#set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -marm -mabi=aapcs-linux -mhard-float -mfloat-abi=hard -mlittle-endian -mtls-dialect=gnu2 -funsafe-math-opti>
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mlittle-endian -funsafe-math-optimizations")
```
I tried this cmake line with dma enabled:
`cmake -DADAFRUIT_HX8357D_PITFT=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DARMV8A=ON ..`
Tried running the binary from an SSH login and nothing seems to happen and I get this error:
```
# DISPLAY=:0 ./fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
Bus error (core dumped)
```
I tried this to disable dma to see if there is any difference:
`cmake -DADAFRUIT_HX8357D_PITFT=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DARMV8A=ON -DUSE_DMA_TRANSFERS=OFF ..`
and now, the display flashes but still does not display anything and I get this error:
```
# DISPLAY=:0 ./fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Initializing display
Creating SPI task thread
InitSPI done
Bus error (core dumped)
```

juj, 2020-10-24:
Hard to say - fbcp-ili9341 has only been tested against 32-bit Raspbian, other OSes are a bit of a lucky scenario if they work. Not sure what is happening there. If you are on 64-bit OS, then maybe some of the memory mapping stuff is completely different, and causes the crash. I'm afraid you'll need to debug through it on your own to figure this one out.

There are some other issues reported against Arch Linux in this bug tracker, maybe some users from those issues may be able to help.. (have not seen a Bus error show up though)

rzr, 2021-04-03:
What about pi3+ on aarch64 as reference device ?