juj/fbcp-ili9341 #177: ST7735R 1.8 Inch 12x160 Display Flickering

frietvet600, 2020-12-13:
**ST7735R Display Flickering Problem.** The screen goes bright every time there's a major(?) frame update. then if the frame doesn't change the screen is dimmer. I noticed that highering the -DSPI_BUS_CLOCK_DIVISOR amount made it seem less bad. Although I haven't gone further than changing it to 80. For if needed my cmake settings are: **-DST7735R=ON -DGPIO_TFT_DATA_CONTROL=5 -DGPIO_TFTF_RESET_PIN=6 -DSPI_BUS_CLOCK_DIVISOR=80 -DSTATISTICS=0 -**DUSE_DMA_TRANSFERS=OFF -DDISPLAY_SWAP_BGR=ON                                                  It isn't a huge problem but its annoying for playing games. Any help is appreciated, this is my first GitHub post so any advise on how to improve this post will help :).

juj, 2020-12-13:
Any chance you'd be able to capture a video (with statistics on), and take a close capture of the display?

frietvet600, 2020-12-14:
> Any chance you'd be able to capture a video (with statistics on), and take a close capture of the display?

The phone Camera wouldn't focus but I did notice when there's no blinking the speed is 3.3mbps and when it does blink around 500kbps

juj, 2020-12-14:
What if you uncomment this line?

https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/config.h#L102

Or this line? (remove the uncommenting of the previous line when testing)

https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/config.h#L111

I wonder if it is a voltage thing, or an interfacing thing.

Other random things you can try is to comment out this line:

https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/st7735r.cpp#L37

And/or this line:

https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/st7735r.cpp#L101

frietvet600, 2020-12-14:
Sadly none of them have seemed to work.

juj, 2020-12-14:
hmm, then I am starting to run out of ideas I'm afraid. Worth checking if any other display drivers avoid the issue, and also scope the voltage lines to check that power supply is adequate, just in case.

frietvet600, 2020-12-14:
ok if i cant get it working it isnt a big problem