juj/fbcp-ili9341 #179: I cant get my 128x128 ssd1351 screen module to work properly

jaleck0, 2020-12-20:
I am currenlty trying to get this screen to work:
https://www.waveshare.com/1.5inch-rgb-oled-module.htm

my cmake code that i used was:

cmake -DSSD1351=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=24 -DSPI_BUS_CLOCK_DIVISOR=30 -STATISTICS=0  ..

juj, 2020-12-21:
That looks good, nothing stands out on the command line. If no other troubleshooting items help, try verifying the display using a different driver to double check physical connection.

ozett, 2020-12-21:
images of wiring could be helpful,
maybe also some information about used hardware/software ?

jaleck0, 2020-12-21:
i get a screen but in /etc/rc.local i put a line of code above the exit 0 that constantly reboots the thing

sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &

jaleck0, 2020-12-21:
and when i simply boot it with:
./fbcp-ili9341
it only uses the lower part of the screen

jaleck0, 2020-12-22:
When i start the compiled software it will only show a screen of 128x96 pixels

ozett, 2020-12-22:
may the developer can help?
may you tried to set the hmdi screen to the oled size?
do this in /boot/config.txt
see readme.md for details..

jaleck0, 2020-12-22:
hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1

maybe i need to so something in the ssd1351.h or ssd1351.cpp before compiling

jaleck0, 2020-12-22:
This is now in my config.txt:

hdmi_group=2
hdmi_mode=87
hdmi_cvt=128 128 60 1 0 0 0
hdmi_force_hotplug=1

maybe i need to so something in the ssd1351.h or ssd1351.cpp before compiling

jaleck0, 2020-12-22:
apperently i had to change a certain value within ssd1351.cpp from 95 to 127 but now the screen has rotated 90 degrees and its mirrored

juj, 2020-12-23:
It reads like you got it going with some pixels at least.

Reading https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/ssd1351.cpp#L36 it suggests there is a vertical swap directive in there. Try changing the `0x10` to `0x00`, or maybe to `0x12`.

Datasheet is available here: https://www.newhavendisplay.com/app_notes/SSD1351.pdf . It has the following about addressing modes:

[image]
[image]

which suggest that if `0x10` is not producing the right addressing for your display, then `0x00` or `0x12` should do it.

jaleck0, 2020-12-24:
Thanks that seemed to work my last question is if there is a way to reduce ghosting and screen tearing

juj, 2020-12-27:
See the documentation section about Screen Tearing at https://github.com/juj/fbcp-ili9341#about-tearing.

Ghosting is an effect of the voltages and refresh rate timings that the panel is driven at. Some displays have specific controls for operating voltages, others do not. The datasheet for SSD1351 above does list some voltage related commands, you can try playing around with those to see if any of them has an effect.