juj/fbcp-ili9341 #180: My OLED is not working

dh928, 2021-01-04:
I followed the instructions here: https://github.com/juj/fbcp-ili9341/blob/master/README.md
And I was trying to connect the Adafruit OLED 128x128 1.5' board to my Raspberry Pi 4B+, but it is just not working. The display seemed like it never turned on. I don't know what is happening. I had it working before on the exact same setting but I don't know why it is not working this time. I checked the wiring and everything. 

I followed this:

sudo apt-get install cmake
cd ~
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
mkdir build
cd build
cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=27 -DGPIO_TFT_RESET_PIN=22 -DSPI_BUS_CLOCK_DIVISOR=30 -DARMV8A=ON -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=OFF -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=OFF -DSTATISTICS=2 -DUSE_DMA_TRANSFERS=OFF -DDISPLAY_INVERT_COLORS=OFF -DDISPLAY_ROTATE_180_DEGREES=ON ..
make -j

and I made sure to edit /etc/rc.local to make sure it launches at startup.
But it is not working at all.

A few months ago, I used the exact same setting and it worked. This time I also tried using different -DSPI_BUS_CLOCK_DIVISOR ranging from 20 to 60. I tried all the even numbers between them. It is just not working. I'm not sure what is going on

juj, 2021-01-09:
Do you have a link to the display? The `-DILI9341=ON` option will definitely not work without modifications on a 128x128 display, because it assumes a 320x240 ILI9341 controller based display.

juj, 2021-01-09:
If it is this one https://www.adafruit.com/product/1431, try using `-DSSD1351=ON` instead of `-DILI9341=ON` option (delete CMakeCache.txt when changing that option)